SICKAG/sick_scan #106: Port 2112 or 2111 is not created.

staycoolish, 2020-10-09:
Hello, I am trying to connect to MRS1000 via ethernet, and could not do it. I created a new connection in Ubuntu, set ip to 192.168.1.24 (the camera has 192.168.1.9), and netmask to 255.255.255.0, left gateway empty. The problem is, port 2111 or 2112 is not created. When I write nmap 192.168.1.24 , only 22/tcp is returned. Similarly, if I write 192.168.1.9 , only 22/tcp is returned, as seen in the figure. Do you have any idea why that happens? By the way, in Windows Sopas is working flawlessly, and I would like to fix the problem in Ubuntu, because finally we will connect the lidar to a computer that won't have Windows, but only Linux. Thank you.

[image]

michael1309, 2020-10-14:
Could you solve this problem? Maybe this is a firewall problem. It seems that this is not a driver problem.

staycoolish, 2020-10-14:
Thanks for the reply. No, the problem still exists but I think the reason is different. I don't know much about IP addresses. I know that we should **define a separate IP** , but I guess I **duplicated the IP address (i.e., defined another )of the computer** somehow. Let me show you what I got: 
[image]
[image]
[image]

I think I might be missing a valid IP address, what do you think? If I write nmap for **192.168.1.9** (IP of computer) and **192.168.1.24** (I defined it) , I get the exact same result, **only 1 port available: 22/tcp**. I also verified that firewall is inactive.
[image]

If I should create a new IP, could you briefly describe what I should do? I try to add another connection by **+ button** at wired, but I think that just replicates the same connection. Also, I guess if there are more than 1 connection for the same MAC address, only 1 of them is activated.

staycoolish, 2020-11-03:
At the bottom of IPv4 configurations, there is an option: 
**Use this connection only for resources on its network.** 
Enabling that solved the problem.

closed 2020-11-03 by staycoolish