SICKAG/sick_scan #107: Cmake error

schleptech, 2020-10-14:
Getting attached error. Can you please help with why catkin_make is not working correctly? Following your directions. 
[image]

michael1309, 2020-10-14:
Did you start this command on the Raspberry?

Please ensure, that the directory
```
/home/pi/ros_catkin_ws/src
```
just contains `sick_scan` for your test.

michael1309, 2020-10-14:
Maybe there are more projects installed in parallel. These projects can contain a package.xml without catkin-dependencies.

schleptech, 2020-10-18:
You were correct - I had not removed all the original projects from installing ROS originally. I removed all the projects except for sick_scan. This worked. Thank you

closed 2020-10-18 by schleptech