SICKAG/sick_scan #110: min_angle and max_angle are not set

HannesBachter, 2021-01-22:
When I start my NAV310 with the provided [launch file](https://github.com/SICKAG/sick_scan/blob/master/launch/sick_nav_3xx.launch), the values for `min_angle` and `max_angle` are not considered. 
The min-/max-values in the `/scan`-topic are always set to: 
```
...
angle_min: 3.1415927410125732
angle_max: -3.128502607345581
...
```
The parameters are set correctly:
```
$ rosparam get /sick_nav_3xx/min_ang 
-3.141592653589793
$ rosparam get /sick_nav_3xx/max_ang
3.141592653589793
```
I can also see in the terminal window where I started the launchfile:
```
...
[ INFO] [1611309329.700527714]: MIN_ANG (after command verification):   -3.142 [rad] -180.000 [deg]
[ INFO] [1611309329.700556413]: MAX_ANG (after command verification):    3.142 [rad]  180.000 [deg]
...
```

michael1309, 2021-01-30:
We will check this during next week  after setting up the NAV310 as test device.

michael1309, 2021-02-04:
Hello Hannes,

we were able to reproduce the behavior you describe with our NAV 310 Firmware V 1.50.
To further investigate this issue, we still have to clarify a few questions with the development department.
A subsequent filtering of the PointCLoud2 data on your side is currently probably the fastest solution and a good workaround.

HannesBachter, 2021-02-05:
Thank you for the information - I found a temporary workaround to work with

michael1309, 2021-03-14:
Close by finding a workaround.

closed 2021-03-14 by michael1309

HannesBachter, 2021-03-16:
We're using a **temporary** workaround for the moment, which is fine for testing but we would require a proper fix for this issue.
Please reopen this issue again