SICKAG/sick_scan #111: sick_scan on windows

cax94, 2021-01-27:
Hi,
i'am new to ROS on windows.
I'am trying to install sick driver sick_scan for an MRS6000 on windows. I've installed ros realese noetic, catkin_tools and visual studio 2019. When i'am run catkin_make command in the catkin workspace it seems all going well when at a certain point it fails giving me this error for 6 times:

_cl : Command line error D8021 '/Wno-format-overflow' not valid
ninja: build stopped: subcommand failed.
Invoking "ninja -j8 -v" failed_
If i have to provide you other information tell me and i immediately try to write in that thread. 
Thanks in advance

michael1309, 2021-01-30:
I think you have run into a general ROS problem here. This is not directly related to sick_scan. We know this kind of error from the Windows environment.  I recommend you to test ROS under Linux. You can use e.g. WSL for this under Windows (see e.g. https://jack-kawell.com/2019/06/24/ros-wsl1/ ).
or you use a virtual machine (e.g. VmWare) together with e.g. Ubuntu.

closed 2021-02-04 by michael1309

cax94, 2021-02-04:
Hi micheal
Thanks for the help, it is not what I really need but through the WSL I solved my problems.
Thanks!!!