SICKAG/sick_scan #112: the paremeter of  the time_increment

chance2006, 2021-01-30:
Dear sir:
I received a message " WARN] [1612000119.987551113]: The time_increment, scan_time and angle_increment values reported by the scanner are inconsistent! Expected time_increment: 0.000185185, reported time_increment: 0.000061728. Perhaps you should set the parameter time_increment to the expected value. This message will print every 60 seconds.
"when I executed "roslanunch sick_scan sick_tim_5xx.launch".
I couldn't find time_increment in the sick_tim_5xx.launch.Would you kindly tell me that how can I change the value of time_increment?  
Thanks very much!

Best regards!

michael1309, 2021-01-30:
Thanks for your hint. We will check this during next week. 
Could you send us the starting sequence of the roslaunch process? What type of Lidar do you use?

michael1309, 2021-02-04:
Hello @chance2006

unfortunately we could not reproduce the error you described.
Are you using the latest version of the sick_scan software?
Which firmware does your scanner use?
What does your launchfile and the startlog look like ?

The calculation of the expected timing can be found here:
https://github.com/SICKAG/sick_scan/blob/40d320545bad4761cf901ddd4aa83f897a4197b8/driver/src/sick_generic_parser.cpp#L764-L781

In this lines you can change the maximum allowed deviation to suppress the warning.

best regards

Michael

chance2006, 2021-02-04:
Dear micheal:
thanks for your reply  I used the tim561  with v4.02，I didn't use a pc.My computers are Raspberry 3b and 4b.
I will test  your method ASAP.Thanks again!
Best Regards！

michael1309, 2021-03-14:
Close due to inactivity and given support.

closed 2021-03-14 by michael1309