SICKAG/sick_scan #113: free(): invalid pointer - Issue when launching LMS511 20100

sebasgra39, 2021-02-03:
Dear Sick-Team,

I have some hard times trying to launch our LMS 511. Before posting the error message from launching, some general info:
The Device has just got the latest firmware update and I'm using the repo on the devel branch. Pining the scanner (currently static IP) is also no issue and I can find it with the generic device finder. 

On an additional windows system, I can also monitor data in the SOPAS ET. 

Within SOPAS, I have the following settings: 
[image]

_When launching the scanner, i get the following errors:_ 

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /sick_lms_5xx/ang_res: 0.25
 * /sick_lms_5xx/frame_id: cloud
 * /sick_lms_5xx/hostname: 192.168.7.81
 * /sick_lms_5xx/intensity: False
 * /sick_lms_5xx/intensity_resolution_16bit: False
 * /sick_lms_5xx/max_ang: 3.228859116
 * /sick_lms_5xx/min_ang: -0.087266463
 * /sick_lms_5xx/port: 2112
 * /sick_lms_5xx/range_max: 80.0
 * /sick_lms_5xx/scan_freq: 25.0
 * /sick_lms_5xx/scanner_type: sick_lms_5xx
 * /sick_lms_5xx/timelimit: 5
 * /sick_lms_5xx/use_binary_protocol: True

NODES
  /
    sick_lms_5xx (sick_scan/sick_generic_caller)

auto-starting new master
process[master]: started with pid [18409]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to c578c1a2-663c-11eb-82a5-2cd444b4ca07
process[rosout-1]: started with pid [18420]
started core service [/rosout]
process[sick_lms_5xx-2]: started with pid [18427]
[ INFO] [1612369677.226346552]: sick_generic_caller V. 1.7.4
[ INFO] [1612369677.226402825]: Program arguments: /home/messnix/sigg_ws/devel/lib/sick_scan/sick_generic_caller
[ INFO] [1612369677.226416410]: Program arguments: __name:=sick_lms_5xx
[ INFO] [1612369677.226437963]: Program arguments: __log:=/home/messnix/.ros/log/c578c1a2-663c-11eb-82a5-2cd444b4ca07/sick_lms_5xx-2.log
[ INFO] [1612369677.240773128]: Found sopas_protocol_type param overwriting default protocol:
[ INFO] [1612369677.241646416]: Binary protocol activated
[ INFO] [1612369677.241691036]: Start initialising scanner [Ip: 192.168.7.81] [Port: 2112]
[ INFO] [1612369677.277021868]: Publishing laserscan-pointcloud2 to cloud
[ INFO] [1612369677.283174759]: Parameter setting for <active_echo: 0>
[ INFO] [1612369677.484305527]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
[ERROR] [1612369683.002582307]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ INFO] [1612369683.002765868]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1612369683.003071962]: Receiving: <STX><ETX>
[ INFO] [1612369683.003264301]: SOPAS Communication -Error unexpected Sopas Answer for request <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>

[ INFO] [1612369683.003766000]: Sending  : <STX>sMN SetAccessMode 3 F4724744<ETX>
[ INFO] [1612369683.010835206]: Receiving: <STX>sAN SetAccessMode 1<ETX>
[ INFO] [1612369683.211335496]: Sending  : <STX>sWN EIHstCola 1<ETX>
[ INFO] [1612369683.212667895]: Receiving: <STX>sWA EIHstCola<ETX>
[ INFO] [1612369683.212834923]: Failed to init scanner Error Code: 1
Waiting for timeout...
If the communication mode set in the scanner memory is different from that used by the driver, the scanner's communication mode is changed.
This requires a restart of the TCP-IP connection, which can extend the start time by up to 30 seconds. There are two ways to prevent this:
1. [Recommended] Set the communication mode with the SOPAS ET software to binary and save this setting in the scanner's EEPROM.
2. Use the parameter "use_binary_protocol" to overwrite the default settings of the driver.
[... 70 lines cut ...]
[ INFO] [1612369702.958668133]: MIN_ANG (after command verification):   -1.658 [rad]  -95.000 [deg]
[ INFO] [1612369702.958741223]: MAX_ANG (after command verification):    1.658 [rad]   95.000 [deg]
[ INFO] [1612369702.958795266]: LMS 5xx detected overwriting output channel flag ID
[ INFO] [1612369702.958845290]: LMS 5xx detected overwriting resolution flag (only 8 bit supported)
[ INFO] [1612369702.959012167]: Sending  : <STX><STX><STX><STX><Len=0032>sWN LMDscandatacfg 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x01 CRC:<0x43>
[ INFO] [1612369702.960259903]: Receiving: <STX>sWA LMDscandatacfg <ETX>
[ INFO] [1612369702.960445108]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMDscandatacfg CRC:<0x67>
[ INFO] [1612369702.961715904]: Receiving: <STX>sRA LMDscandatacfg \x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x01<ETX>
[ INFO] [1612369702.963706750]: Sending  : <STX><STX><STX><STX><Len=0036>sMN mLMPsetscancfg 0x2b 0x02 0x05 0x00 0x00 0x20 0x01 0x20 0x2b 0x02 0x05 0x00 0x00 0x20 0x00 0x20 0x00 CRC:<0x52>
[ INFO] [1612369702.964989673]: Receiving: <STX>sFA\x00\x08<ETX>
[ERROR] [1612369702.965150219]: Error Sopas answer mismatch Error unexpected Sopas Answer for request <STX><STX><STX><STX><Len=0036>sMN mLMPsetscancfg 0x2b 0x02 0x05 0x00 0x00 0x20 0x01 0x20 0x2b 0x02 0x05 0x00 0x00 0x20 0x00 0x20 0x00 CRC:<0x52>Answer= >>>sFA\x00\x08<<<

[ INFO] [1612369702.965337980]: Sending  : <STX><STX><STX><STX><Len=0014>sRN LMPscancfg CRC:<0x63>
[ INFO] [1612369702.966618892]: Receiving: <STX>sRA LMPscancfg \x00\x00\x09\xc4\x00\x01\x00\x00\x09\xc4\xff\xff\x3c\xb0\x00\x1c\x3a\x90<ETX>
[ INFO] [1612369702.966854606]: Sending  : <STX><STX><STX><STX><Len=0016>sMN LMCstartmeas CRC:<0x68>
[ INFO] [1612369702.968122665]: Receiving: <STX>sAN LMCstartmeas \x00<ETX>
[ INFO] [1612369702.968317896]: Sending  : <STX><STX><STX><STX><Len=0007>sMN Run CRC:<0x19>
[ INFO] [1612369702.969580747]: Receiving: <STX>sAN Run \x01<ETX>
[ INFO] [1612369702.969777058]: Sending  : <STX><STX><STX><STX><Len=0017>sEN LMDscandata 0x01 CRC:<0x33>
[ INFO] [1612369702.971032351]: Receiving: <STX>sEA LMDscandata \x01<ETX>
free(): invalid pointer

_Have you got any idea what I could possible test out?_
_We have a rather old version of the scanner and I had to adjust the max. and min angle within the launch file, since SOPAS ET does not allow to use min angles below -5 for our model. We also have the 20100 Version, what might be worth mentioning too.
BR and many thanks in advance!_

michael1309, 2021-02-04:
Hello Sebsastian,

Thank you very much for your comment.
Currently the master branch with version 1.8.0 was some commits ahead of the devel branch (but after your hint we merged the code into the devel branch).
You should therefore use the code from the master branch for your scanner.
I could not reproduce the error described by you with the current software version 1.8.0. so this code should work for you.

I hope this answer helps you.

Best regards Michael

sebasgra39, 2021-02-04:
Hello Michael,

thanks for your fast answer and kind reply. Unfortunately, switching to the master branch did not help. I still get the following list of Errors: 

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /sick_lms_5xx/ang_res: 0.25
 * /sick_lms_5xx/frame_id: cloud
 * /sick_lms_5xx/hostname: 192.168.7.81
 * /sick_lms_5xx/intensity: False
 * /sick_lms_5xx/intensity_resolution_16bit: False
 * /sick_lms_5xx/max_ang: 3.228859116
 * /sick_lms_5xx/min_ang: -0.087266463
 * /sick_lms_5xx/port: 2112
 * /sick_lms_5xx/range_max: 80.0
 * /sick_lms_5xx/scan_freq: 25.0
 * /sick_lms_5xx/scanner_type: sick_lms_5xx
 * /sick_lms_5xx/timelimit: 5
 * /sick_lms_5xx/use_binary_protocol: True

NODES
  /
    sick_lms_5xx (sick_scan/sick_generic_caller)

auto-starting new master
process[master]: started with pid [5358]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 854014cc-66fb-11eb-8abd-2cd444b4ca07
process[rosout-1]: started with pid [5369]
started core service [/rosout]
process[sick_lms_5xx-2]: started with pid [5376]
[ INFO] [1612451603.325108760]: sick_generic_caller V. 1.8.0
[ INFO] [1612451603.325149595]: Program arguments: /home/messnix/sigg_ws/devel/lib/sick_scan/sick_generic_caller
[ INFO] [1612451603.325159646]: Program arguments: __name:=sick_lms_5xx
[ INFO] [1612451603.325181341]: Program arguments: __log:=/home/messnix/.ros/log/854014cc-66fb-11eb-8abd-2cd444b4ca07/sick_lms_5xx-2.log
[ INFO] [1612451603.334093399]: Found sopas_protocol_type param overwriting default protocol:
[ INFO] [1612451603.334887373]: Binary protocol activated
[ INFO] [1612451603.334908660]: Start initialising scanner [Ip: 192.168.7.81] [Port: 2112]
[ INFO] [1612451603.363687429]: Publishing laserscan-pointcloud2 to cloud
[ INFO] [1612451603.374809284]: Parameter setting for <active_echo: 0>
[ INFO] [1612451603.575745740]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
[ INFO] [1612451603.579415548]: Receiving: <STX>sAN SetAccessMode \x01<ETX>
[ INFO] [1612451603.779873084]: Sending  : <STX><STX><STX><STX><Len=0015>sWN EIHstCola 0x01 CRC:<0x09>
[ INFO] [1612451603.781233792]: Receiving: <STX>sWA EIHstCola <ETX>
[ INFO] [1612451603.981670380]: Sending  : <STX><STX><STX><STX><Len=0019>sRN FirmwareVersion CRC:<0x24>
[ INFO] [1612451603.983038609]: Receiving: <STX>sRA FirmwareVersion \x00\x09\x56\x31\x2e\x38\x33\x2e\x31\x20\x20<ETX>
[ INFO] [1612451604.183470305]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[ INFO] [1612451604.184812197]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[ INFO] [1612451604.385290500]: Sending  : <STX><STX><STX><STX><Len=0010>sRN ODoprh CRC:<0x41>
[ INFO] [1612451604.386640935]: Receiving: <STX>sRA ODoprh \x00\x00\x3b\xe7<ETX>
[ INFO] [1612451604.588628117]: Sending  : <STX><STX><STX><STX><Len=0010>sRN ODpwrc CRC:<0x52>
[ INFO] [1612451604.589980881]: Receiving: <STX>sRA ODpwrc \x00\x00\x03\x0f<ETX>
[ INFO] [1612451604.791943735]: Sending  : <STX><STX><STX><STX><Len=0016>sRN LocationName CRC:<0x55>
[ INFO] [1612451604.793306461]: Receiving: <STX>sRA LocationName \x00\x06\x50\x41\x4d\x34\x2e\x30<ETX>
[ INFO] [1612451604.795091244]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMPoutputRange CRC:<0x5e>
[ INFO] [1612451604.796388040]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x09\xc4\xff\xff\x3c\xb0\x00\x1c\x3a\x90<ETX>
[ INFO] [1612451604.796547476]: Angle resolution of scanner is 0.25000 [deg]  (in 1/10000th deg: 0x9C4)
[ INFO] [1612451604.796611168]: [From:To] -5.00000 [deg] to 185.00000 [deg] (in 1/10000th deg: from 0xFFFF3CB0 to 0x1C3A90)
[ INFO] [1612451604.796661576]: MIN_ANG:   -0.087 [rad]   -5.000 [deg]
[ INFO] [1612451604.796703007]: MAX_ANG:    3.229 [rad]  185.000 [deg]
[ INFO] [1612451604.796862807]: Sending  : <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 0x00 0x09 0xc4 0x00 0x0c 0xf8 0x50 0x00 0x29 0xf6 0x30 CRC:<0xfc>
[ INFO] [1612451604.798120277]: Receiving: <STX>sFA\x00\x04<ETX>
[ERROR] [1612451604.798264522]: Error Sopas answer mismatch Error unexpected Sopas Answer for request <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 0x00 0x09 0xc4 0x00 0x0c 0xf8 0x50 0x00 0x29 0xf6 0x30 CRC:<0xfc>Answer= >>>sFA\x00\x04<<<

[ INFO] [1612451604.798475873]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMPoutputRange CRC:<0x5e>
[... 134 lines cut ...]

[ INFO] [1612451604.887195309]: Sending  : <STX><STX><STX><STX><Len=0012>sRN field047 CRC:<0x1e>
[ INFO] [1612451604.888443559]: Receiving: <STX>sFA\x00\x03<ETX>
[ERROR] [1612451604.888583743]: Error Sopas answer mismatch Error unexpected Sopas Answer for request <STX><STX><STX><STX><Len=0012>sRN field047 CRC:<0x1e>Answer= >>>sFA\x00\x03<<<

[ INFO] [1612451604.888820565]: Sending  : <STX><STX><STX><STX><Len=0032>sWN LMDscandatacfg 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x01 CRC:<0x43>
[ INFO] [1612451604.890063628]: Receiving: <STX>sWA LMDscandatacfg <ETX>
[ INFO] [1612451604.890255590]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMDscandatacfg CRC:<0x67>
[ INFO] [1612451604.891532115]: Receiving: <STX>sRA LMDscandatacfg \x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x01<ETX>
[ INFO] [1612451604.893740332]: Sending  : <STX><STX><STX><STX><Len=0037>sMN mLMPsetscancfg 0x00 0x00 0x09 0xc4 0x00 0x01 0x00 0x00 0x09 0xc4 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 CRC:<0x52>
[... cut ...]
[ INFO] [1612451605.083422201]: Sending  : <STX><STX><STX><STX><Len=0007>sMN Run CRC:<0x19>
[ INFO] [1612451605.084725822]: Receiving: <STX>sAN Run \x01<ETX>
[ INFO] [1612451605.084960633]: Sending  : <STX><STX><STX><STX><Len=0017>sEN LMDscandata 0x01 CRC:<0x33>
[ INFO] [1612451605.086233288]: Receiving: <STX>sEA LMDscandata \x01<ETX>
free(): invalid pointer
[sick_lms_5xx-2] process has died 

_I also tried setting the binary mode to false, but this also not lead to success. I hope to hear from You soon.
BR
Sebastian_

michael1309, 2021-02-04:
There are a few issues:

- first correct min_ang and max_ang to ROS convention. The ROS driver uses ROS-coordinate convention (X-axis straight ahead). SICK uses for must lidars the "SICK" convention (X-Axis points to the right)
```
<param name="min_ang" type="double" value="-1.658062789"/>
<param name="max_ang" type="double" value="1.658062789"/>
```
- For crash debugging follow the gdb instruction given in  https://github.com/SICKAG/sick_scan/issues/85
- "Field ..." warnings can be ignored

michael1309, 2021-03-14:
Closes due to inactivity.

closed 2021-03-14 by michael1309