SICKAG/sick_scan #115: Sick_scan on Windows 10

Mecatropabien, 2021-03-04:
Hello,

I'm working on a TIM240 and I would like to use sick_scan on Windows 10. I had some trouble first because some librairies (inet.h, socket.h) are for Unix.
Sick send me a link to a closed issue which can help me : https://github.com/SICKAG/sick_scan_base/issues/6

I saw your recommandations and I installed vcpkg, boost and pthread. But I don't understand the next step about build.bat.
which build.bat ? Where I can find it ?

[image]

Thank you in advance for your response.
Best regards,
Julien REY.

michael1309, 2021-03-08:
sick_scan is currently not fully supported for Windows. If you want to use sick_scan (application for ROS) and not sick_scan_base (native c++ application), then I recommend you to use Windows Subsystem for Linux (WSL2) https://docs.microsoft.com/en-us/windows/wsl/install-win10

Mecatropabien, 2021-03-09:
Hi, thank you very much for your reply. I installed WSL2 on my computer. Very easy to install it and ROS. As we are on Linux, the driver can be compile without any problem.

closed 2021-03-09 by Mecatropabien