SICKAG/sick_scan #116: Will this code support interlaced mode for LMS5xx

themightyoarfish, 2021-03-11:
According to https://www.sick.com/us/en/detection-and-ranging-solutions/2d-lidar-sensors/lms5xx/lms511-20100-pro/p/p216240, the LMS5xx PRO scanner supports an interlaced mode to achieve higher resolutions. The telegram listing does not mention this functionality for the LMS, only others.

Will this mode be supported in the future?

michael1309, 2021-03-11:
Hello @themightyoarfish 

Thank you for your request.
We performed a test and can confirm your assumption. The angle interleaved resolutions 0.083 ° and 0.042 ° are currently not supported for LMS 5xx lidar.
If you wish the support for this feature we recommend to send a request to Sick https://supportportal.sick.com .

Best regards

michael1309, 2021-03-14:
Close due to "currently not supported".

closed 2021-03-14 by michael1309