SICKAG/sick_scan #117: Ethernet cable not detected TiM581-2050101

shubhankarpataskar, 2021-03-12:
Hello! 
I installed ROS Noetic on Ubuntu 20.04 and set the firewall status as active but I'm facing an issue when I try to connect the LiDAR. I get 
[image] 

What is the best way to tackle this, I tried the same on my raspberry pi 4 (Buster 64 bit) it connects but
sudo apt-get install ros-noetic-desktop fails due to 
libgl-dev package hence it cannot install ros-noetic-sick-scan

michael1309, 2021-03-14:
Can you ping the lidar under 169.254.7.26? 
Could you also disable the firewall for a couple of minutes and try it again?
Did you set up a permanent ip using SOPAS ET?
As far as I know 169.254.x.x is used for DHCP etc. (see https://packetlife.net/blog/2008/sep/24/169-254-0-0-addresses-explained/ )
**I really recommend to disable DHCP and use a fix ip address for communication with the lidar (like 192.168.0.100 for the pc or raspberry) and 192.168.0.1 for the lidar.**

shubhankarpataskar, 2021-03-16:
I can ping under the given IP address. 
Also I did disable the firewall and checked 
I'm working on setting up a permanent IP using SOPAS ET.

michael1309, 2021-03-16:
OK. Here are some more hints for you.

Please set this Ethernet adapter to a static IP address in the address space 192.168.0.xxx (we recommend 192.168.0.100). 
Then please try to ping the Lidar at the address 192.168.0.1. 
If this is successful, also use netcat to check if the Lidar can be reached on port 2112 by running the following command:

nc -z -v -w5 192.168.0.1 2112 

For more information, see e.g. 

                https://github.com/SICKAG/sick_scan/blob/master/doc/ipconfig/ipconfig.md

For setting up a static IP, you can follow the standard documentation for Ubuntu, etc.

shubhankarpataskar, 2021-03-16:
I tried setting up a static IP

[image]

but the SOPAS ET won't read the device for 192.168.XX.XXX IPs

michael1309, 2021-03-18:
We are glad to see that the Lidar is pingable under Windows. If you can't reach the device with ping under Linux, this is not a problem of the lidar. In this case it is likely that the network infrastructure is not set up correctly. Maybe you are using DHCP or there is a routing problem. We recommend you to reserve the address space 192.168.0.xxx for the communication with the Lidar. Please contact your network administrator and ask him to set the local Ethernet adapter (i.e. the one installed in your PC) to e.g. 192.168.0.100 with netmask 255.255.255.0. Then you can reach the Lidar also under the address 192.168.0.1. Of course, no other Ethernet adapter may use this address space, otherwise there will be conflicts in the IP traffic.

closed 2021-03-18 by michael1309