SICKAG/sick_scan #121: Changing min_ang and max_ang leads to incorrect data

themightyoarfish, 2021-04-17:
I have an LMS511 20100 PRO.

When I start the driver with `min_ang:=-1`, `max_ang:=1.5`, I obtain this scan:

[image]

When I start the driver with `min_ang:=-1.658062789`, `max_ang:=1.658062789` for the full angular range, I obtain this scan:

[image]

Clearly, the coordinate system gets rotated instead of the angular range getting limited.
The launch code is this (ignore the defaults, all gets set):
```
<arg name="hostname" default="192.168.0.1" />
    <arg name="device" default="lms" />
    <!-- -135 to 135 deg -->
    <arg name="min_ang" default="-2.35619449" />
    <arg name="max_ang" default="2.35619449" />
    <arg name="range_min" default="0.2" />
    <arg name="range_max" default="100" />
    <!-- Only for lms -->
    <arg name="ang_res" default="0.3333" />
    <arg name="scan_freq" default="50" />

    <node if="$(eval arg('device') == 'lms')" name="sick_lms_5xx" pkg="sick_scan"
        type="sick_generic_caller"  respawn="false" output="screen">

        <param name="scanner_type" type="string" value="sick_lms_5xx" />
        <param name="min_ang" type="double" value="$(arg min_ang)" />
        <param name="max_ang" type="double" value="$(arg max_ang)" />
        <param name="use_binary_protocol" type="bool" value="true" />
        <param name="range_min" type="double" value="$(arg range_min)" />
        <param name="range_max" type="double" value="$(arg range_max)" />
        <param name="intensity" type="bool" value="True" />
        <param name="hostname" type="string" value="$(arg hostname)" />
        <param name="ang_res" type="double" value="$(arg ang_res)" />
        <param name="scan_freq" type="double" value="$(arg scan_freq)" />
    </node>
```

The scanner acknowledges the configured ranges. The scan resolution does not impact this behaviour. What could this be caused by?

themightyoarfish, 2021-04-17:
Tested with current master branch.

michael1309, 2021-04-18:
Good point. 

I have measured the angle of rotation between the two drawings. This corresponds quite exactly to the start angle difference of 0.658062789 rad. Probably, therefore, the new start angle of -1.0 rad is not correctly considered. We will investigate this further in the next few days. Until then, we recommend you to use the default setup and to filter not needed angle values in the application.

themightyoarfish, 2021-04-18:
That was my hypothesis, but the code was not so obvious to me. I appreciate any forthcoming fix for this.

michael1309, 2021-04-22:
We have repeated the test in our laboratory. You will find the results of the test attached. We performed the test against the LMS511-10100. Unfortunately, we could not reproduce the described behavior. The pictures are attached. We have tested it against the current head revision. The launch file we tested with is also attached. You can launch it with roslaunch sick_scan t.launch if you save this file into the launch directory before. In our test we directly specified the start angle and the stop angle.

If you still have problems, please copy the complete launch log into the ticket. Good luck!
[image]
[image]
[image]
```
<?xml version="1.0"?>
<!--
*****************************************************************************
* Launch Example File for SICK LMS 511                                      *
*****************************************************************************
-->
<!-- start and stop angle is given in [rad] -->
<!-- 
Check IP-address, if you scanner is not found after roslaunch.
-->
<launch>
    <arg name="hostname" default="192.168.0.151"/>
    <!-- launch file for the lms_5xx series -->
    <node name="sick_lms_5xx" pkg="sick_scan" type="sick_generic_caller" respawn="false" output="screen">
        <!--<param name="robot_description" command="$(find xacro)/xacro.py '$(find sick_scan)/urdf/example.urdf.xacro'" />-->

        <!-- default values: -->
        <!--
          <param name="min_ang" type="double" value="0.0" />
          <param name="max_ang" type="double" value="1.658062789" />
          <param name="intensity" type="bool" value="True" />
          <param name="skip" type="int" value="0" />
          <param name="frame_id" type="str" value="laser" />
          <param name="time_offset" type="double" value="-0.001" />
          <param name="publish_datagram" type="bool" value="False" />
          <param name="subscribe_datagram" type="bool" value="false" />
          <param name="device_number" type="int" value="0" />
          <param name="range_min" type="double" value="0.05" />
        -->
        <param name="frame_id" type="str" value="cloud"/>
        <param name="scanner_type" type="string" value="sick_lms_5xx"/>
        <!-- -95° -->
        <param name="min_ang" type="double" value="-0.0"/>
        <!-- 95° -->
        <param name="max_ang" type="double" value="1.658062789"/>
        <param name="range_max" type="double" value="100.0"/>
        <param name="intensity" type="bool" value="true"/>
        <param name="intensity_resolution_16bit" type="bool" value="false"/>
        <param name="hostname" type="string" value="$(arg hostname)"/>
        <param name="port" type="string" value="2112"/>
        <param name="timelimit" type="int" value="5"/>
        <param name="use_binary_protocol" type="bool" value="true"/>
        <!-- possible angular resolutions are 1,0.6667,0.5,0.3333,0.25,0.1667
         possible scan frequencies are 25,35,50,75,100 Hz
         Make sure to check in the manual whether the desired scanning frequency
         and angular resolution combination is available.
         The parameters scan_freq and ang_res must always be used together.
        <param name="ang_res" type="double" value="0.1667" />
        <param name="scan_freq" type="double" value="25" />
         -->
        <param name="start_services" type="bool" value="True"/> <!-- start ros service for cola commands -->
        <param name="activate_lferec" type="bool" value="True"/> <!-- activate field monitoring by lferec messages -->
        <param name="activate_lidoutputstate" type="bool" value="True"/> <!-- activate field monitoring by lidoutputstate messages -->
        <param name="activate_lidinputstate" type="bool" value="True"/> <!-- activate field monitoring by lidinputstate messages -->

    </node>
</launch>
```

And here the screenshot showing the output for this setup:
[image]

michael1309, 2021-05-11:
Close due to inactivity.

closed 2021-05-11 by michael1309