SICKAG/sick_scan #122: Rviz error

shubhankarpataskar, 2021-05-05:
I face an error while i run the 
rosrun rviz rviz 
The status shows error once the topic /scan is selected. 

[image]
 
Is there any solution you guys can help me with?

rostest, 2021-05-06:
sick_scan publishes with frame id "cloud". Please select cloud in the rviz "Fixed Frame" field to solve the error:
[image]

shubhankarpataskar, 2021-05-06:
Thank you!
worked perfectly.
Also how did you get the detection field to be displayed?

closed 2021-05-06 by shubhankarpataskar

rostest, 2021-05-06:
Thanks for your feedback. Just add a MarkerArray in rviz to display the detection field:
[image]
You can find more details about field monitoring for LMS1xx, LMS5xx, TiM7xx and TiM7xxS in https://github.com/SICKAG/sick_scan/blob/master/doc/field_monitoring_extensions.md

shubhankarpataskar, 2021-05-06:
I do not get the option, under 
/sick_tim (im using TIM 581). I get all these options:
[image]

rostest, 2021-05-06:
Detection fields and field monitoring are only supported for LMS1xx, LMS5xx, TiM7xx and TiM7xxS, not for TIM 581.

shubhankarpataskar, 2021-05-06:
Okay, Thank you!