SICKAG/sick_scan #126: Field monitoring for MRS1000

honyu-sick, 2021-05-18:
Dear Support Team,

Based on the link : https://github.com/SICKAG/sick_scan/blob/master/doc/field_monitoring_extensions.md

The function of the Field monitoring were only for LMS1xx, LMS5xx, TiM7xx and TiM7xxS. 

May I verify if the MRS1104 model have this feature?  My understanding from this feature can be activate in SOPAS but I am unsure if it is available in ROS.

Kindly need your assistance to advise. Thanks!

Regards
Hon

michael1309, 2021-05-19:
Hello,

thank you for your interest. I will forward this request to the support team at SICK for prioritization. Within SICK support for then decide on a possible implementation. If you find the opportunity to integrate this feature, we will be happy to integrate it as a pull request.

Best regards

michael1309, 2021-06-16:
Hi,

After consultation with SICK management, this feature will not be implemented for the moment, because there is no request for this feature at the official SICK support. 
If you are interested in this feature for commercial projects etc. we recommend to send this request directly to the official SICK support with further explanation of the project environment on your side.

closed 2021-06-16 by michael1309