SICKAG/sick_scan #127: Number of points per layer for MRS6xxx differs from specifications

themightyoarfish, 2021-05-23:
On page 16 of https://cdn.sick.com/media/docs/0/40/540/operating_instructions_mrs6000_en_im0076540.pdf it says that 924 points are emitted per layer (120° / 0.13° + 1), but the ROS messaged produced by the driver only contains 922.

Where does this discrepancy come from?

michael1309, 2021-05-23:
The start and end angles on the MRS6124 have been minimally adjusted to output 922 points instead of 924 points. Therefore, your observation is correct and intended.

closed 2021-05-23 by michael1309

themightyoarfish, 2021-05-23:
What does this mean for the radial coordinates of the points? Does the azimuth angle start at -60° and go in 922 steps of 0.13°? Or is everything shifted somehow?

michael1309, 2021-05-25:
There is no internal angle shift. The settings only result in the first and last angle entry not being transferred. The calculation of the pointcloud data etc. is therefore also correct.