SICKAG/sick_scan #128: SickScanFieldMonSingleton::parseAsciiDatagram not implemented

ksoltan, 2021-05-27:
Hi there!
We recently started seeing the following error when starting up our SICK lidars (we have 2, a front and back):
```
/sick_scan_front [1622070201.163093577] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():756]: Receiving: <STX>sRA field000 3F800000 00000000 D05 FFF92230 2 1 1 0 0 0 0 1 8 Field001 0<ETX>
/sick_scan_front [1622070201.163157133] [ERROR] [SickScanFieldMonSingleton::parseAsciiDatagram():237]: SickScanFieldMonSingleton::parseAsciiDatagram not implemented.
/sick_scan_front [1622070201.163220650] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():750]: Sending  : <STX>sRN field001<ETX>
/sick_scan_back [1622070201.163719476] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():756]: Receiving: <STX>sRA field000 3F800000 00000000 D05 FFF92230 2 1 1 0 0 0 0 1 8 Field001 0<ETX>
/sick_scan_back [1622070201.163773447] [ERROR] [SickScanFieldMonSingleton::parseAsciiDatagram():237]: SickScanFieldMonSingleton::parseAsciiDatagram not implemented.
/sick_scan_back [1622070201.163824216] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():750]: Sending  : <STX>sRN field001<ETX>
/sick_scan_front [1622070201.165527294] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():756]: Receiving: <STX>sRA field001 3F800000 00000000 D05 FFF92230 2 2 1 0 0 0 0 1 8 Field002 0<ETX>
```
This is only happening on our systems that have version 1.10.1 of sick_scan installed. Our other system is running version 1.7.8, and is not producing this error.

Do you have any idea what the problem may be? Is there something that changed between these two versions?
Thank you very much!

michael1309, 2021-05-27:
We have also implemented field monitoring in the meantime. However, field monitoring only supports binary SOPAS communication. We recommend using SOPAS binary communication by default anyway for performance reasons and low latency. Please switch the communication of the lidars via SOPAS ET permanently to binary communication. Then this problem will not occur anymore. Alternatively you can ignore these error messages as long as everything else works as usual.

closed 2021-05-27 by michael1309

ksoltan, 2021-05-28:
@michael1309 We changed the lidars to use binary communication, and we're still seeing the same error. Do you have any other suggestions?

michael1309, 2021-05-29:
Maybe you set the protocol to Ascii. Please check your launch file for an entry 
like:
```
        <param name="use_binary_protocol" type="bool" value="true"/>
```
If this parameter is set to `false`, the lidar starts with the (unwished) ASCII protocol. 
If you run into further problems, we recommend to send the whole startup sequence include your launch command and description and your lidar type (see https://github.com/SICKAG/sick_scan#support ).

ksoltan, 2021-06-09:
Thank you, that fixed the original error. We're running into a timeout error now, but I'll open up a [new issue](https://github.com/SICKAG/sick_scan/issues/130) for that.