SICKAG/sick_scan #130: Getting timeout error when using binary protocol

ksoltan, 2021-06-09:
Hi again,
After changing the protocol to binary (from [this issue](https://github.com/SICKAG/sick_scan/issues/128)), we're seeing the following error message:
```
[SickScanCommonTcp::readWithTimeout():650]: no answer received after 5000 ms. Maybe sopas mode is wrong.
```
We increased the timelimit from 5 to 6 ( `<param name="timelimit" type="int" value="6" />`), and we no longer see the error. 

However, why is this happening in the first place? This [previous issue](https://github.com/SICKAG/sick_scan/issues/6) suggested changing the time limit to 30,000ms, but this was supposed to be fixed in the next release?

Thank you for your time!

michael1309, 2021-06-09:
Hi,

could you send the whole start protocol for a better understanding?
What is your OS and PC platform?

ksoltan, 2021-06-09:
Hi,
We're running Ubuntu 20.04 on a Stealth mini-PC. Here's our launch file (where time-limit was changed to 6 and the errors went away):
```
<?xml version="1.0"?>
<!--
     Launch File for starting up SICK TIM 781S sensors.
-->

<launch>
  <arg name="front_sick_ip" default="$(optenv SICK_FRONT_IP)" />
  <arg name="front_cloud_topic" default="front_cloud" />
  <arg name="front_frame_id" default="/laser_front" />

  <arg name="back_sick_ip" default="$(optenv SICK_BACK_IP)" />
  <arg name="back_cloud_topic" default="back_cloud" />
  <arg name="back_frame_id" default="/laser_back" />

  <!-- <param name="robot_description" command="$(find xacro)/xacro.py '$(find sick_scan)/urdf/example.urdf.xacro'" /> -->
  <node name="sick_scan_front" pkg="sick_scan" type="sick_generic_caller" respawn="false" output="screen">
    <param name="scanner_type" type="string" value="sick_tim_7xxS" />

    <param name="min_ang" type="double" value="-2.35619449" />
    <param name="max_ang" type="double" value="2.35619449" />

    <param name="use_binary_protocol" type="bool" value="false" />
    <param name="range_max" type="double" value="100.0" />
    <param name="intensity" type="bool" value="True" />

    <param name="hostname" type="string" value="$(arg front_sick_ip)" />
    <param name="cloud_topic" type="string" value="$(arg front_cloud_topic)"/>
    <param name="frame_id" type="str" value="$(arg front_frame_id)" />

    <param name="port" type="string" value="2112" />
    <param name="timelimit" type="int" value="5" />

    <remap from="scan" to="scan_front" />
    <remap from="cloud" to="cloud_front" />
  </node>

  <node name="sick_scan_back" pkg="sick_scan" type="sick_generic_caller" respawn="false" output="screen">
    <param name="scanner_type" type="string" value="sick_tim_7xxS" />

    <param name="min_ang" type="double" value="-2.35619449" />
    <param name="max_ang" type="double" value="2.35619449" />

    <param name="use_binary_protocol" type="bool" value="false" />
    <param name="range_max" type="double" value="100.0" />
    <param name="intensity" type="bool" value="True" />

    <param name="hostname" type="string" value="$(arg back_sick_ip)" />
    <param name="cloud_topic" type="string" value="$(arg back_cloud_topic)"/>
    <param name="frame_id" type="str" value="$(arg back_frame_id)" />

    <param name="port" type="string" value="2112" />
    <param name="timelimit" type="int" value="5" />

    <remap from="scan" to="scan_back" />
    <remap from="cloud" to="cloud_back" />
  </node>
</launch>
```

michael1309, 2021-06-10:
Just for clarification: Is your problem gone by changing the time-limit to 6? Is the problem solved?

ksoltan, 2021-06-16:
Oh, I misunderstood. Below is the output.
Yes, the problem goes away when we increase the time-limit to 10. What changes were made that we now have to increase the timeout? Also, is there a way to suppress some of the logging output? We're seeing significantly more logs now with the change to binary protocol.

Here is the problematic startup output:
```
[${node}] [1623801417.660230999] [ INFO] [main():188]: sick_generic_caller V. 1.10.1
[${node}] [1623801417.660276849] [ INFO] [main():196]: Program arguments: /opt/ros/noetic/lib/sick_scan/sick_generic_caller
[${node}] [1623801417.660292903] [ INFO] [main():196]: Program arguments: scan:=scan_front
[${node}] [1623801417.660307950] [ INFO] [main():196]: Program arguments: cloud:=cloud_front
[${node}] [1623801417.660322715] [ INFO] [main():196]: Program arguments: __name:=sick_scan_front
[${node}] [1623801417.660335875] [ INFO] [main():196]: Program arguments: __log:=/home/katerina/beta-borg/artifacts/5de12672-ce35-11eb-aaf1-71277970c60e/sick_scan_front-2.log
process[sick_scan_back-3]: started with pid [144240]
process[zone_activation_node-4]: started with pid [144252]
[${node}] [1623801417.671342237] [ INFO] [main():188]: sick_generic_caller V. 1.10.1
[${node}] [1623801417.671402081] [ INFO] [main():196]: Program arguments: /opt/ros/noetic/lib/sick_scan/sick_generic_caller
[${node}] [1623801417.671420700] [ INFO] [main():196]: Program arguments: scan:=scan_back
[${node}] [1623801417.671434997] [ INFO] [main():196]: Program arguments: cloud:=cloud_back
[${node}] [1623801417.671449530] [ INFO] [main():196]: Program arguments: __name:=sick_scan_back
[${node}] [1623801417.671466733] [ INFO] [main():196]: Program arguments: __log:=/home/katerina/beta-borg/artifacts/5de12672-ce35-11eb-aaf1-71277970c60e/sick_scan_back-3.log
process[obstacle_node-5]: started with pid [144259]
[/sick_scan_front] [1623801417.679298879] [ INFO] [string)():283]: Found sopas_protocol_type param overwriting default protocol:
[/sick_scan_front] [1623801417.681030445] [ INFO] [string)():286]: Binary protocol activated
[/sick_scan_front] [1623801417.681064317] [ INFO] [string)():327]: Start initialising scanner [Ip: 192.168.1.203] [Port: 2112]
[/sick_scan_back] [1623801417.697288222] [ INFO] [string)():283]: Found sopas_protocol_type param overwriting default protocol:
[/sick_scan_back] [1623801417.701732177] [ INFO] [string)():286]: Binary protocol activated
[/sick_scan_back] [1623801417.701758598] [ INFO] [string)():327]: Start initialising scanner [Ip: 192.168.1.204] [Port: 2112]
[/sick_scan_front] [1623801417.738717504] [ INFO] [SickScanCommon::SickScanCommon():392]: Publishing laserscan-pointcloud2 to front_cloud
[/sick_scan_front] [1623801417.743765595] [ INFO] [SickScanCommon::init_scanner():1221]: Parameter setting for <active_echo: 0>
[/sick_scan_back] [1623801417.745727252] [ INFO] [SickScanCommon::SickScanCommon():392]: Publishing laserscan-pointcloud2 to back_cloud
[/sick_scan_back] [1623801417.748949688] [ INFO] [SickScanCommon::init_scanner():1221]: Parameter setting for <active_echo: 0>
[/sick_scan_front] [1623801417.944589290] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():750]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0x6f 0xd6 0x2c 0x05 CRC:<0xa6>
[/sick_scan_back] [1623801417.949522962] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():750]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0x6f 0xd6 0x2c 0x05 CRC:<0xa6>
[/sick_scan_front] [1623801423.539640950] [ERROR] [SickScanCommonTcp::readWithTimeout():650]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[/sick_scan_front] [1623801423.539752527] [ INFO] [SickScanCommonTcp::sendSOPASCommand():765]: sendSOPASCommand: no full reply available for read after 5000 ms
[/sick_scan_front] [1623801423.540003202] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():756]: Receiving: <STX><ETX>
[/sick_scan_front] [1623801423.540115424] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():761]: SOPAS Communication -Error unexpected Sopas Answer for request <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0x6f 0xd6 0x2c 0x05 CRC:<0xa6>

[/sick_scan_front] [1623801423.540281192] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():750]: Sending  : <STX>sMN SetAccessMode 3 6FD62C05<ETX>
[/sick_scan_back] [1623801423.543996119] [ERROR] [SickScanCommonTcp::readWithTimeout():650]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[/sick_scan_back] [1623801423.544147689] [ INFO] [SickScanCommonTcp::sendSOPASCommand():765]: sendSOPASCommand: no full reply available for read after 5000 ms
[/sick_scan_back] [1623801423.544344593] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():756]: Receiving: <STX><ETX>
[/sick_scan_back] [1623801423.544450919] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():761]: SOPAS Communication -Error unexpected Sopas Answer for request <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0x6f 0xd6 0x2c 0x05 CRC:<0xa6>

[/sick_scan_back] [1623801423.544613086] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():750]: Sending  : <STX>sMN SetAccessMode 3 6FD62C05<ETX>
[/sick_scan_front] [1623801423.547022473] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():756]: Receiving: <STX>sAN SetAccessMode 1<ETX>
[/sick_scan_back] [1623801423.551382647] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():756]: Receiving: <STX>sAN SetAccessMode 1<ETX>
[... cut ...]
[/sick_scan_back] [1623801882.076538114] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():750]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[/sick_scan_back] [1623801882.079060082] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():756]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[/sick_scan_front] [1623801882.264484500] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():750]: Sending  : <STX><STX><STX><STX><Len=0010>sRN ODoprh CRC:<0x41>
[/sick_scan_front] [1623801882.267073474] [ INFO] [SickScanCommon::sendSopasAndCheckAnswer():756]: Receiving: <STX>sRA ODoprh \x00\x00\x07\x40<ETX>
```

ksoltan, 2021-06-16:
Ok, we verified that the lidar was set to binary by default, through SOPAS ET.|
[image]

We continued to get the `Found sopas_protocol_type param overwriting default protocol` message until we commented out in the launch file `<param name="use_binary_protocol" type="bool" value="true" />`.
Like you predicted, it looks like the timeout isn't being hit anymore.

The verbosity of the logs did not change with this fix, but it looks like this is something we'll have to patch ourselves, correct?

rostest, 2021-06-18:
The verbosity level of ros nodes can be set in the launch file, f.e. by
```
<launch>
  <env name="ROSCONSOLE_CONFIG_FILE" value="/tmp/rosconsole_loglevel_warn.conf"/>
```
This overwrites the default configuration in `$ROS_ROOT/config/rosconsole.config` (which logs all INFO messages by default). To suppress informational messages, the log level can be set to `log4j.logger.ros=WARN` in file `/tmp/rosconsole_loglevel_warn.conf`. See http://wiki.ros.org/rosconsole#Configuration for examples and details. Just be aware that informational messages can help with diagnostics.

closed 2021-06-24 by michael1309