SICKAG/sick_scan #131: Add an option to filter scan by intensity

ManMan88, 2021-06-17:
It would be great to add a feature which will enable to filter the scan by the intensity values, similar to the "min_intensities" parameter in the sick_safetyscanners package.

michael1309, 2021-06-24:
Can you tell a little more about this requirement? 
Are you talking about the Pointcloud2 message containing fewer messages in this case? 
Specifically, should the values that are below the threshold then be removed?

Basically, the filter functionality by intensity, distance range, angle range, etc. can be performed in the application.  

We also consider this to be the best procedure in principle, because in this case we can ensure that the Pointcloud2 values can be passed in a fixed structure (concretely: matrix structure), so that one can assume here with most lidars that a distance value is always measured at the same matrix position within this matrix always in the same spatial direction.

ManMan88, 2021-06-27:
I am talking about the /scan topic (LaserScan message). And yes, intensities below the threshold should be removed.
The way it works in the sick_safetyscanners package is that the data points which are below the threshold are filled with an invalid value (I don't remember if it is filled with inf, or nan or something else). Then the relevant packages which use this data would automatically ignore the those data points. This way the fixed structure of the LaserScan message is not being changed.

Yes, the filter can be implemented in the application, still it would be more convenient if it would be part of the driver.

michael1309, 2021-06-27:
OK. We understand your point. We will integrate it as part of restructuring work over the next few weeks. After reviewing the documentation for sick_safetyscanners, these values should be set to infinity.

rostest, 2021-07-13:
Option  min_intensity is added in the last release 1.11.0. If min_intensity > 0 and intensity < min_intensity, then the range is set to inf in LaserScan messages. Option `min_intensity` can be set in the launchfile.

closed 2021-07-13 by rostest