SICKAG/sick_scan #133: occationally scan data not appeared

clobot-jade, 2021-06-30:
Hi, Im using this pkg in this env.
melodic
2 TIM781S and merged with ira merger ( front , rear ) 

occationally, front lidar scan data is not appeared in several seconds while rear lidar is working.
what im do for debugging is below 
1. tried ping to scanner : is working 
2. tried nc -z -v ... : is working 
3. sequence in rostopic echo is same to rear lidar, which means all lidar received anyway 

well, it seems like mutex curruption or tcp data queue matter. 
how could fix it? if want a rosbag file. i can send for you

michael1309, 2021-07-01:
Hi,

In my experience, this is likely to be a data capture issue. In any case I recommend you (if not already set) to use SOPAS-Binary instead of SOPAS-ASCII, because here the data rate is lower.

In addition I recommend you the following measures, in order to determine the source of the problems:
a) Swap the cables to the lidars among each other
b) Swap the power supply to the lidars among each other
c) Exchange of the network cables among each other
d) avoidance of interconnection of other components (like switch etc.) or
 hardware upgrade of these devices

If it is not a hardware problem (e.g. cable), I recommend to check if there are other software components using the network interface.

If there are still problems, I recommend to use Wireshark and see if there are any communication problems reported. 

For this you can export the recorded network traffic as text (see screenshot). In the generated text file you can then search for the entry LMDscandata. This marks the beginning of a new scan message. You can then compare the order of the timestamps of these packets to see if there were any failures.

[image]

closed 2021-07-01 by michael1309

clobot-jade, 2021-07-22:
well, it still not working occationally.
remind you that the issue is front lidar only, rear lidar is not in case.

1. 
it turns out that we use binary protocol already. 

2. 
i did your workaround.
a) Swap the cables to the lidars among each other
- cannot test for wiring issue .. i will next time 

b) Swap the power supply to the lidars among each other
- the lidars share their power line. did you see that specific lidar occur fault in same smps?

c) Exchange of the network cables among each other
- i swaped lan port each other. but the case is same.
- didnt test swapping wire

d) avoidance of interconnection of other components (like switch etc.) or
hardware upgrade of these devices
- well, there is no network switch. pc have 6 port
- exchanged front lidar with new one, but case is same

but is it normal that scan data is not appear while ping is working? 
some times it is not work forever and cpu useage rise to 100%.. i dont think it is just wiring issue
is there any debug process?
or how can i send you log data?