SICKAG/sick_scan #134: LMS5xx driver problem

MitchFeike, 2021-07-15:
Scanner: LMS511

Startup log:
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://md3ffn2c.ad001.siemens.net:45455/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.11
 * /sick_lms_5xx/activate_lferec: True
 * /sick_lms_5xx/activate_lidinputstate: True
 * /sick_lms_5xx/activate_lidoutputstate: True
 * /sick_lms_5xx/frame_id: cloud
 * /sick_lms_5xx/hostname: 192.168.10.171
 * /sick_lms_5xx/intensity: True
 * /sick_lms_5xx/intensity_resolution_16bit: False
 * /sick_lms_5xx/max_ang: 1.658062789
 * /sick_lms_5xx/min_ang: -1.658062789
 * /sick_lms_5xx/port: 2112
 * /sick_lms_5xx/range_max: 65.0
 * /sick_lms_5xx/scanner_type: sick_lms_5xx
 * /sick_lms_5xx/start_services: True
 * /sick_lms_5xx/timelimit: 5
 * /sick_lms_5xx/use_binary_protocol: True

NODES
  /
    sick_lms_5xx (sick_scan/sick_generic_caller)

ROS_MASTER_URI=http://localhost:11311

process[sick_lms_5xx-1]: started with pid [30351]
[ INFO] [1626347531.550135821]: sick_generic_caller V. 1.8.0
[ INFO] [1626347531.550248190]: Program arguments: /home/mitch/projects/masterarbeit/code/ros_ws/devel/lib/sick_scan/sick_generic_caller
[ INFO] [1626347531.550281880]: Program arguments: __name:=sick_lms_5xx
[ INFO] [1626347531.550309330]: Program arguments: __log:=/home/mitch/.ros/log/1b361448-e55c-11eb-8aeb-3d4852e86f19/sick_lms_5xx-1.log
[ INFO] [1626347531.594615431]: Found sopas_protocol_type param overwriting default protocol:
[ INFO] [1626347531.597152949]: Binary protocol activated
[ INFO] [1626347531.597222795]: Start initialising scanner [Ip: 192.168.10.171] [Port: 2112]
[ INFO] [1626347531.690032621]: Publishing laserscan-pointcloud2 to cloud
[ INFO] [1626347531.704049900]: Parameter setting for <active_echo: 0>
[ INFO] [1626347531.906265807]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
[ INFO] [1626347531.911510561]: Receiving: <STX>sAN SetAccessMode \x01<ETX>
[ INFO] [1626347532.112156107]: Sending  : <STX><STX><STX><STX><Len=0015>sWN EIHstCola 0x01 CRC:<0x09>
[ INFO] [1626347532.113581625]: Receiving: <STX>sWA EIHstCola <ETX>
[ INFO] [1626347532.314188745]: Sending  : <STX><STX><STX><STX><Len=0019>sRN FirmwareVersion CRC:<0x24>
[ INFO] [1626347532.315643176]: Receiving: <STX>sRA FirmwareVersion \x00\x09\x56\x31\x2e\x35\x30\x2e\x38\x20\x20<ETX>
[ INFO] [1626347532.516246075]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[ INFO] [1626347532.517791347]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[ INFO] [1626347532.718450179]: Sending  : <STX><STX><STX><STX><Len=0010>sRN ODoprh CRC:<0x41>
[ INFO] [1626347532.719907349]: Receiving: <STX>sRA ODoprh \x00\x00\x1f\xdd<ETX>
[ INFO] [1626347532.922309950]: Sending  : <STX><STX><STX><STX><Len=0010>sRN ODpwrc CRC:<0x52>
[ INFO] [1626347532.923755944]: Receiving: <STX>sRA ODpwrc \x00\x00\x08\xdf<ETX>
[ INFO] [1626347533.126077465]: Sending  : <STX><STX><STX><STX><Len=0016>sRN LocationName CRC:<0x55>
[ INFO] [1626347533.127521356]: Receiving: <STX>sRA LocationName \x00\x0b\x6e\x6f\x74\x20\x64\x65\x66\x69\x6e\x65\x64<ETX>
[ INFO] [1626347533.129639147]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMPoutputRange CRC:<0x5e>
[ INFO] [1626347533.130952100]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x09\xc4\xff\xff\x3c\xb0\x00\x1c\x3a\x90<ETX>
[ INFO] [1626347533.131139409]: Angle resolution of scanner is 0.25000 [deg]  (in 1/10000th deg: 0x9C4)
[ INFO] [1626347533.131213705]: [From:To] -5.00000 [deg] to 185.00000 [deg] (in 1/10000th deg: from 0xFFFF3CB0 to 0x1C3A90)
[ INFO] [1626347533.131273320]: MIN_ANG:   -1.658 [rad]  -95.000 [deg]
[ INFO] [1626347533.131347081]: MAX_ANG:    1.658 [rad]   95.000 [deg]
[ INFO] [1626347533.131546554]: Sending  : <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 0x00 0x09 0xc4 0xff 0xff 0x3c 0xb0 0x00 0x1c 0x3a 0x90 CRC:<0x8d>
[ INFO] [1626347533.132835587]: Receiving: <STX>sWA LMPoutputRange <ETX>
[ INFO] [1626347533.133068977]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMPoutputRange CRC:<0x5e>
[... 73 lines cut ...]
[ INFO] [1626347533.198874902]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMDscandatacfg CRC:<0x67>
[ INFO] [1626347533.200235619]: Receiving: <STX>sRA LMDscandatacfg \x01\x00\x01\x00\x00\x00\x00\x00\x00\x00\x01\x00\x01<ETX>
[ INFO] [1626347533.204000731]: sick_lms_5xx: activating field monitoring by lferec messages
[ INFO] [1626347533.205349638]: sick_lms_5xx: activating field monitoring by lidoutputstate messages
[ INFO] [1626347533.206626210]: sick_lms_5xx: activating field monitoring by lidinputstate messages
[ INFO] [1626347533.206893139]: Sending  : <STX><STX><STX><STX><Len=0016>sMN LMCstartmeas CRC:<0x68>
[ INFO] [1626347533.208267008]: Receiving: <STX>sAN LMCstartmeas \x00<ETX>
[ INFO] [1626347533.208600285]: Sending  : <STX><STX><STX><STX><Len=0007>sMN Run CRC:<0x19>
[ INFO] [1626347533.209963379]: Receiving: <STX>sAN Run \x01<ETX>
[ INFO] [1626347533.210250037]: Sending  : <STX><STX><STX><STX><Len=0017>sEN LMDscandata 0x01 CRC:<0x33>
[ INFO] [1626347533.211613831]: Receiving: <STX>sEA LMDscandata \x01<ETX>
[ INFO] [1626347533.211904441]: Sending  : <STX><STX><STX><STX><Len=0012>sEN LFErec 0x01 CRC:<0x42>
[ INFO] [1626347533.213276258]: Receiving: <STX>sEA LFErec \x01<ETX>
[ INFO] [1626347533.213568272]: Sending  : <STX><STX><STX><STX><Len=0020>sEN LIDoutputstate 0x01 CRC:<0x50>
[ INFO] [1626347533.216077827]: Receiving: <STX>sEA LIDoutputstate \x01<ETX>
[ INFO] [1626347533.216367982]: Sending  : <STX><STX><STX><STX><Len=0019>sEN LIDinputstate 0x01 CRC:<0x39>
[ INFO] [1626347533.218908998]: Receiving: <STX>sEA LIDinputstate \x01<ETX>
[ INFO] [1626347533.225249336]: SickScanServices: ros services initialized
free(): invalid pointer
[sick_lms_5xx-1] process has died [pid 30351, exit code -6, cmd /home/mitch/projects/masterarbeit/code/ros_ws/devel/lib/sick_scan/sick_generic_caller __name:=sick_lms_5xx __log:=/home/mitch/.ros/log/1b361448-e55c-11eb-8aeb-3d4852e86f19/sick_lms_5xx-1.log].
log file: /home/mitch/.ros/log/1b361448-e55c-11eb-8aeb-3d4852e86f19/sick_lms_5xx-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

MitchFeike, 2021-07-15:
process[sick_lms_5xx-1]: started with pid [30978]
[ INFO] [1626348398.673415042]: sick_generic_caller V. 1.8.0
[ INFO] [1626348398.673527091]: Program arguments: /home/mitch/projects/masterarbeit/code/ros_ws/devel/lib/sick_scan/sick_generic_caller
[ INFO] [1626348398.673557197]: Program arguments: __name:=sick_lms_5xx
[ INFO] [1626348398.673585359]: Program arguments: __log:=/home/mitch/.ros/log/1b361448-e55c-11eb-8aeb-3d4852e86f19/sick_lms_5xx-1.log
[ INFO] [1626348398.718521513]: Found sopas_protocol_type param overwriting default protocol:
[ INFO] [1626348398.721139462]: Binary protocol activated
[ INFO] [1626348398.721214055]: Start initialising scanner [Ip: 192.168.10.171] [Port: 2112]
[ INFO] [1626348398.814676369]: Publishing laserscan-pointcloud2 to cloud
[ INFO] [1626348398.830424703]: Parameter setting for <active_echo: 0>
[ INFO] [1626348399.032505673]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
[ INFO] [1626348399.037870067]: Receiving: <STX>sAN SetAccessMode \x01<ETX>
[ INFO] [1626348399.238587725]: Sending  : <STX><STX><STX><STX><Len=0015>sWN EIHstCola 0x01 CRC:<0x09>
[ INFO] [1626348399.240081506]: Receiving: <STX>sWA EIHstCola <ETX>
[ INFO] [1626348399.440766717]: Sending  : <STX><STX><STX><STX><Len=0019>sRN FirmwareVersion CRC:<0x24>
[ INFO] [1626348399.442266854]: Receiving: <STX>sRA FirmwareVersion \x00\x09\x56\x31\x2e\x35\x30\x2e\x38\x20\x20<ETX>
[ INFO] [1626348399.642988635]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[ INFO] [1626348399.644497913]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[ INFO] [1626348399.845116779]: Sending  : <STX><STX><STX><STX><Len=0010>sRN ODoprh CRC:<0x41>
[ INFO] [1626348399.846595359]: Receiving: <STX>sRA ODoprh \x00\x00\x1f\xe0<ETX>
[ INFO] [1626348400.048810962]: Sending  : <STX><STX><STX><STX><Len=0010>sRN ODpwrc CRC:<0x52>
[ INFO] [1626348400.050284633]: Receiving: <STX>sRA ODpwrc \x00\x00\x08\xdf<ETX>
[ INFO] [1626348400.252628490]: Sending  : <STX><STX><STX><STX><Len=0016>sRN LocationName CRC:<0x55>
[ INFO] [1626348400.254067067]: Receiving: <STX>sRA LocationName \x00\x0b\x6e\x6f\x74\x20\x64\x65\x66\x69\x6e\x65\x64<ETX>
[ INFO] [1626348400.256174201]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMPoutputRange CRC:<0x5e>
[ INFO] [1626348400.257575803]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x09\xc4\xff\xff\x3c\xb0\x00\x1c\x3a\x90<ETX>
[ INFO] [1626348400.257804932]: Angle resolution of scanner is 0.25000 [deg]  (in 1/10000th deg: 0x9C4)
[ INFO] [1626348400.257897484]: [From:To] -5.00000 [deg] to 185.00000 [deg] (in 1/10000th deg: from 0xFFFF3CB0 to 0x1C3A90)
[ INFO] [1626348400.257991155]: MIN_ANG:   -1.658 [rad]  -95.000 [deg]
[ INFO] [1626348400.258072070]: MAX_ANG:    1.658 [rad]   95.000 [deg]
[ INFO] [1626348400.258296023]: Sending  : <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 0x00 0x09 0xc4 0xff 0xff 0x3c 0xb0 0x00 0x1c 0x3a 0x90 CRC:<0x8d>
[ INFO] [1626348400.259652040]: Receiving: <STX>sWA LMPoutputRange <ETX>
[ INFO] [1626348400.259912886]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMPoutputRange CRC:<0x5e>
[ INFO] [1626348400.261295372]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x09\xc4\xff\xff\x3c\xb0\x00\x1c\x3a\x90<ETX>
[ INFO] [1626348400.261517507]: Angle resolution of scanner is 0.25000 [deg]  (in 1/10000th deg: 0x9C4)
[ INFO] [1626348400.264580655]: MIN_ANG (after command verification):   -1.658 [rad]  -95.000 [deg]
[ INFO] [1626348400.264686199]: MAX_ANG (after command verification):    1.658 [rad]   95.000 [deg]
[ INFO] [1626348400.264785737]: LMS 5xx detected overwriting output channel flag ID
[ INFO] [1626348400.264868712]: LMS 5xx detected overwriting resolution flag (only 8 bit supported)
[ INFO] [1626348400.264958230]: Reading safety fields
[ INFO] [1626348400.265259342]: Sending  : <STX><STX><STX><STX><Len=0012>sRN field000 CRC:<0x1d>
[ INFO] [1626348400.267797732]: Receiving: <STX>sRA field000 @\x00\x00\x00\x00\x00\x00\x00\x00\x00\x13\x88\xff\xff\x3c\xb0\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00<ETX>
[ INFO] [1626348400.268132006]: Sending  : <STX><STX><STX><STX><Len=0012>sRN field001 CRC:<0x1c>
[ INFO] [1626348400.269532108]: Receiving: <STX>sRA field001 @\x00\x00\x00\x00\x00\x00\x00\x00\x00\x13\x88\xff\xff\x3c\xb0\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00<ETX>
[ INFO] [1626348400.269823876]: Sending  : <STX><STX><STX><STX><Len=0012>sRN field002 CRC:<0x1f>
[ INFO] [1626348400.271216002]: Receiving: <STX>sRA field002 @\x00\x00\x00\x00\x00\x00\x00\x00\x00\x13\x88\xff\xff\x3c\xb0\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00<ETX>
[ INFO] [1626348400.271506981]: Sending  : <STX><STX><STX><STX><Len=0012>sRN field003 CRC:<0x1e>
[ INFO] [1626348400.274024648]: Receiving: <STX>sRA field003 @\x00\x00\x00\x00\x00\x00\x00\x00\x00\x13\x88\xff\xff\x3c\xb0\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00<ETX>
[ INFO] [1626348400.274318942]: Sending  : <STX><STX><STX><STX><Len=0012>sRN field004 CRC:<0x19>
[ INFO] [1626348400.276827712]: Receiving: <STX>sRA field004 @\x00\x00\x00\x00\x00\x00\x00\x00\x00\x13\x88\xff\xff\x3c\xb0\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00<ETX>
[ INFO] [1626348400.277141479]: Sending  : <STX><STX><STX><STX><Len=0012>sRN field005 CRC:<0x18>
[... cut ...]
free(): invalid pointer
[sick_lms_5xx-1] process has died [pid 30978, exit code -6, cmd /home/mitch/projects/masterarbeit/code/ros_ws/devel/lib/sick_scan/sick_generic_caller __name:=sick_lms_5xx __log:=/home/mitch/.ros/log/1b361448-e55c-11eb-8aeb-3d4852e86f19/sick_lms_5xx-1.log].
log file: /home/mitch/.ros/log/1b361448-e55c-11eb-8aeb-3d4852e86f19/sick_lms_5xx-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete

MitchFeike, 2021-07-15:
[ INFO] [1626348400.323689979]: Safety fieldset response to "sRN LIDinputstate": \x02\x02\x02\x02\x00\x00\x00\x1esRA LIDinputstate \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00 , active fieldset = 0
[ INFO] [1626348400.324037589]: Safety fieldset 0, pointcounter = [  0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 ]
[ INFO] [1626348400.324289340]: Sending  : <STX><STX><STX><STX><Len=0032>sWN LMDscandatacfg 0x01 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x01 CRC:<0x42>
[ INFO] [1626348400.325617495]: Receiving: <STX>sWA LMDscandatacfg <ETX>
[ INFO] [1626348400.325830967]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMDscandatacfg CRC:<0x67>
[ INFO] [1626348400.327143710]: Receiving: <STX>sRA LMDscandatacfg \x01\x00\x01\x00\x00\x00\x00\x00\x00\x00\x01\x00\x01<ETX>
[ INFO] [1626348400.330403114]: sick_lms_5xx: activating field monitoring by lferec messages
[ INFO] [1626348400.331471305]: sick_lms_5xx: activating field monitoring by lidoutputstate messages
[ INFO] [1626348400.332543268]: sick_lms_5xx: activating field monitoring by lidinputstate messages
[ INFO] [1626348400.332766251]: Sending  : <STX><STX><STX><STX><Len=0016>sMN LMCstartmeas CRC:<0x68>
[ INFO] [1626348400.334112382]: Receiving: <STX>sAN LMCstartmeas \x00<ETX>
[ INFO] [1626348400.334356151]: Sending  : <STX><STX><STX><STX><Len=0007>sMN Run CRC:<0x19>
[ INFO] [1626348400.335702198]: Receiving: <STX>sAN Run \x01<ETX>
[ INFO] [1626348400.335954341]: Sending  : <STX><STX><STX><STX><Len=0017>sEN LMDscandata 0x01 CRC:<0x33>
[ INFO] [1626348400.337301058]: Receiving: <STX>sEA LMDscandata \x01<ETX>
[ INFO] [1626348400.337553470]: Sending  : <STX><STX><STX><STX><Len=0012>sEN LFErec 0x01 CRC:<0x42>
[ INFO] [1626348400.338905062]: Receiving: <STX>sEA LFErec \x01<ETX>
[ INFO] [1626348400.339155406]: Sending  : <STX><STX><STX><STX><Len=0020>sEN LIDoutputstate 0x01 CRC:<0x50>
[ INFO] [1626348400.341659854]: Receiving: <STX>sEA LIDoutputstate \x01<ETX>
[ INFO] [1626348400.341910103]: Sending  : <STX><STX><STX><STX><Len=0019>sEN LIDinputstate 0x01 CRC:<0x39>
[ INFO] [1626348400.343253933]: Receiving: <STX>sEA LIDinputstate \x01<ETX>
[ INFO] [1626348400.349112358]: SickScanServices: ros services initialized
free(): invalid pointer
[sick_lms_5xx-1] process has died [pid 30978, exit code -6, cmd /home/mitch/projects/masterarbeit/code/ros_ws/devel/lib/sick_scan/sick_generic_caller __name:=sick_lms_5xx __log:=/home/mitch/.ros/log/1b361448-e55c-11eb-8aeb-3d4852e86f19/sick_lms_5xx-1.log].
log file: /home/mitch/.ros/log/1b361448-e55c-11eb-8aeb-3d4852e86f19/sick_lms_5xx-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...

michael1309, 2021-07-15:
Hi,

please install the most recent version of the driver and rebuild the ROS node.
BTW: What is your platform?

Best regards

MitchFeike, 2021-07-15:
I already installed the most recent version.

[ INFO] [1626353093.213514438]: Receiving: <STX>sSN LMDscandata \x00\x00\x00\x01\x00\xf5\x86\xf0\x01\x00\xbf\xa0\xc4\xec\x95\x31\xf6\x60\x95\x32\x3d\xaf\x00\x00\x00\x00\x00\x00\x00\x00\x27\x10\x00\x00\x02\x1c\x00\x00\x00\x05\x44\x49\x53\x54\x31\x3f\x80\x00\x00\x00\x00\x00\x00\xff\xff\x3c\xb0\x1a\x0b\x01\x1e\x00\x51\x00\x55\x07\xff\x08\x14\x08\x25\x08\x39\x08\x4b\x08\x64\x08\x79\x08\x89\x08\x9d\x08\xb5\x08\xd8\x08\xed\x09\x02\x09\x20\x09\x41\x09\x64\x09\x80\x09\x96\x09\x9b\x09\xc0\x09\xe3\x0a\x09\x0a\x38\x0a\x5f\x0a\x87\x0a\xb0\x0a\xe2\x0b\x16\x0b\x4a\x0b\x78\x0b\xb1\x0b\xea\x0c\x25\x0c\x69\x0c\xb0\x0d\x25\x0d\x71\x0d\xbd\x0e\x12\x0e\x6e\x0e\xc9\x0f\x31\x0f\x26\x0f\x1e\x0f\x5e\x0f\xbd\x10\x36\x10\x68\x10\x54\x10\x42\x10\x2a\x10\x25\x10\x0c\x10\x01\x0f\xf4\x0f\xe2\x0f\xde\x0f\xcf\x0f\xc6\x0f\xb3\x0f\xb0\x0f\xab\x0f\xa1\x0f\x93\x0f\x9b\x0f\x80\x0f\x6f\x07\x5e\x07\x55\x0f\x61\x0f\x54\x0f\x57\x0a\x8e\x0a\x6e\x0a\x69\x0a\x92\x0a\xa8\x0f\x4d\x0f\x51\x0f\x4d\x0f\x53\x0f\x55\x0f\x5c\x0f\x58\x0f\x58\x0f\x5e\x0f\x5a\x0f\x77\x0f\x87\x0f\x8d\x0f\x95\x0f\x55\x0e\x13\x0e\x03\x0e\x0a\x0e\x1f\x0e\x29\x0e\x31\x0e\x3f\x0e\x46\x0e\x57\x0e\x60\x0e\x6e\x0e\x7e\x0e\x90\x0e\x98\x0e\xa6\x0e\xb6\x0e\xc6\x0e\xd8\x0e\xeb\x0e\xf8\x0f\x0e\x0f\x20\x0f\x36\x0f\x5a\x0f\x61\x0f\x7b\x0f\x93\x0f\xab\x0f\xc8\x0f\xe1\x0f\xf4\x10\x1b\x10\x2d\x10\x44\x0c\xd2\x0c\xa1\x0c\x90\x0c\xb2\x11\x0e\x11\x11\x11\x3d\x11\x66\x11\x8d\x11\xaf\x11\xe7\x12\x0e\x12\x52\x12\x91\x12\xc1\x12\xf5\x13\x2e\x13\x61\x13\x9d\x13\xdd\x14\x1c\x14\x5b\x14\x6b\x14\x46\x13\xac\x13\xb0\x13\x9b\x13\x78\x12\xe8\x13\x11\x12\xec\x12\x92\x12\x05\x12\x20\x12\x50\x12\x1f\x11\xf2\x11\xcc\x11\xab\x11\x82\x11\x5b\x11\x41\x11\x24\x10\xfe\x10\xec\x10\xd0\x10\xb4\x10\xa6\x10\x93\x10\x75\x10\x5a\x10\x44\x10\x32\x10\x1e\x10\x18\x0f\xf7\x0f\xec\x0f\xde\x0f\xd4\x0f\xc6\x0f\xaf\x0f\xa0\x0f\x95\x0f\x8c\x0f\x86\x0f\x7f\x0f\x6f\x0f\x65\x0f\x6e\x0f\x66\x0f\x5e\x0f\x5b\x0f\x51\x0f\x3f\x0f\x3c\x0f\x39\x0f\x34\x0f\x2d\x0b\xaa\x0b\x25\x0b\x13\x0b\x25\x0f\x37\x0f\x37\x0f\x33\x0f\x2c\x0f\x28\x0f\x21\x0f\x2b\x0f\x0c\x0e\xa5\x0e\x8d\x0f\x22\x07\x14\x07\x0a\x07\x07\x07\x03\x07\x0a\x07\x0d\x0f\x18\x0e\xd6\x0f\x1d\x0f\x3b\x0f\x43\x00\x00\x00\x00\xe9\xf7\x00\x00\x0a\x00\x0a\x04\x0a\x10\x0a\x2d\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xe5\x80\x00\x00\xe5\x55\x00\x00\xe5\x44\xe5\x63\x05\x68\x05\x50\x05\x35\x05\x16\x04\xfd\x04\xec\x05\x03\x05\x0a\x05\x35\x05\x56\x05\x48\x05\x47\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x44\x49\x53\x54\x32\x3f\x80\x00\x00\x00\x00\x00\x00\xff\xff\x3c\xb0\x1a\x0b\x01\x1e\x07\xe1\x07\xef\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0f\x91\x0f\x88\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0f\x6d\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10\x8a\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0f\x57\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0f\x6f\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x44\x49\x53\x54\x33\x3f\x80\x00\x00\x00\x00\x00\x00\xff\xff\x3c\xb0\x1a\x0b\x01\x1e\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x [... cut ...] \x00<<<

[ERROR] [1626353093.219073833]: Error activating LIDinputstate messages
[ INFO] [1626353093.225788130]: SickScanServices: ros services initialized
[sick_lms_5xx-2] process has died [pid 14496, exit code -8, cmd /home/mitch/projects/masterarbeit/code/ros_ws/devel/lib/sick_scan/sick_generic_caller __name:=sick_lms_5xx __log:=/home/mitch/.ros/log/720af3e8-e56a-11eb-8153-cf7c73e980f7/sick_lms_5xx-2.log].
log file: /home/mitch/.ros/log/720af3e8-e56a-11eb-8153-cf7c73e980f7/sick_lms_5xx-2*.log

MitchFeike, 2021-07-15:
Launch File:

<?xml version="1.0"?>

<launch>
    <arg name="hostname" default="192.168.10.171"/>
    
    <node name="sick_lms_5xx" pkg="sick_scan" type="sick_generic_caller" respawn="false" output="screen">
        

        
        
        <param name="frame_id" type="str" value="cloud"/>
        <param name="scanner_type" type="string" value="sick_lms_5xx"/>
        
        <param name="min_ang" type="double" value="-1.658062789"/>
        
        <param name="max_ang" type="double" value="1.658062789"/>
        <param name="range_max" type="double" value="100.0"/>
        <param name="intensity" type="bool" value="true"/>
        <param name="intensity_resolution_16bit" type="bool" value="false"/>
        <param name="hostname" type="string" value="$(arg hostname)"/>
        <param name="port" type="string" value="2112"/>
        <param name="timelimit" type="int" value="5"/>
        <param name="use_binary_protocol" type="bool" value="true"/>
        
	<param name="ang_res" type="double" value="0.6667" />
	<param name="scan_freq" type="double" value="100.0" />
        <param name="start_services" type="bool" value="True"/> 
        <param name="activate_lferec" type="bool" value="True"/> 
        <param name="activate_lidoutputstate" type="bool" value="True"/> 
        <param name="activate_lidinputstate" type="bool" value="True"/> 
        <param name="min_intensity" type="double" value="0.0"/> 

    </node>
</launch>

rostest, 2021-07-15:
Thanks for your report. Can you retry with a lower scanning frequency configured in the launchfile? F.e. 
`<param name="scan_freq" type="double" value="25" />`
This might help in case of network or timing problems.

rostest, 2021-07-15:
The latest sick_scan version is "1.10.1", the version used is "1.8.0" according to the log file. Please rebuild sick_scan with the latest version from the master branch.

michael1309, 2021-07-29:
Close due to inactivity. We assume, that switching the version solved this problem.

closed 2021-07-29 by michael1309