SICKAG/sick_scan #135: occationally scan data not appeared (re)

clobot-jade, 2021-07-22:
#133 

it still not working occationally.
remind you that the issue is front lidar only, rear lidar is not in case.

1. 
it turns out that we use binary protocol already. 

2. 
i did your workaround.
a) Swap the cables to the lidars among each other
- cannot test for wiring issue .. i will next time 

b) Swap the power supply to the lidars among each other
- the lidars share their power line. did you see that specific lidar occur fault in same smps?

c) Exchange of the network cables among each other
- i swaped lan port each other. but the case is same.
- didnt test swapping wire

d) avoidance of interconnection of other components (like switch etc.) or
hardware upgrade of these devices
- well, there is no network switch. pc have 6 port
- exchanged front lidar with new one, but case is same

but is it normal that scan data is not appear while ping is working? 
some times it is not work forever and cpu useage rise to 100%.. i dont think it is just wiring issue
is there any debug process?
or how can i send you log data?

michael1309, 2021-07-29:
Hi,
we have discussed this point internally. We are still confused that these problems only occur with the front lidar and not with the rear lidar. Is there anything different about the IP communication or similar? Is the driver hanging up or is it just that no data is being sent? Can you restart the front lidar after stopping the driver without a cold start? Or is it possible that the front lidar is no longer sending data (i.e. the SOPAS messages are no longer sent)? 

What is in any case helpful for further analysis is the transmission of the startup log.

michael1309, 2021-07-29:
For example, you could use Wireshark to check if SOPAS data is still being received even if you are no longer receiving Pointcloud2 data with ROS.

clobot-jade, 2021-08-02:
hi, turns out we use realsense2_camera.. it published tf in 100000 hz. fixed in 50 hz and the case now not appeared.
im sorry there is no chance to check that. the project is now over :(.. maybe next time i will.

michael1309, 2021-08-02:
OK. Thanks for you feedback and good luck.

closed 2021-08-02 by michael1309