SICKAG/sick_scan #136: LMS5xxx driver problem

MitchFeike, 2021-08-03:
`
SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.11
 * /sick_lms_5xx/activate_lferec: True
 * /sick_lms_5xx/activate_lidinputstate: True
 * /sick_lms_5xx/activate_lidoutputstate: True
 * /sick_lms_5xx/frame_id: cloud
 * /sick_lms_5xx/hostname: 192.168.10.171
 * /sick_lms_5xx/intensity: True
 * /sick_lms_5xx/intensity_resolution_16bit: False
 * /sick_lms_5xx/max_ang: 1.658062789
 * /sick_lms_5xx/min_ang: -1.658062789
 * /sick_lms_5xx/min_intensity: 0.0
 * /sick_lms_5xx/port: 2112
 * /sick_lms_5xx/range_max: 100.0
 * /sick_lms_5xx/scanner_type: sick_lms_5xx
 * /sick_lms_5xx/start_services: True
 * /sick_lms_5xx/timelimit: 5
 * /sick_lms_5xx/use_binary_protocol: True

NODES
  /
    sick_lms_5xx (sick_scan/sick_generic_caller)

auto-starting new master
process[master]: started with pid [73925]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 77bb77da-f42a-11eb-a365-8dc58b81f6cb
process[rosout-1]: started with pid [73935]
started core service [/rosout]
process[sick_lms_5xx-2]: started with pid [73942]
[ INFO] [1627974880.699658977]: sick_generic_caller V. 1.10.1
[ INFO] [1627974880.699709412]: Program arguments: /home/mitch/projects/masterarbeit/code/ros_ws/devel/lib/sick_scan/sick_generic_caller
[ INFO] [1627974880.699733828]: Program arguments: __name:=sick_lms_5xx
[ INFO] [1627974880.699742605]: Program arguments: __log:=/home/mitch/.ros/log/77bb77da-f42a-11eb-a365-8dc58b81f6cb/sick_lms_5xx-2.log
[ INFO] [1627974880.712020108]: Found sopas_protocol_type param overwriting default protocol:
[ INFO] [1627974880.712799863]: Binary protocol activated
[ INFO] [1627974880.712824622]: Start initialising scanner [Ip: 192.168.10.171] [Port: 2112]
[ INFO] [1627974880.737925913]: Publishing laserscan-pointcloud2 to cloud
[ INFO] [1627974880.747352103]: Parameter setting for <active_echo: 0>
[ INFO] [1627974880.949266824]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
[ INFO] [1627974880.955417806]: Receiving: <STX>sAN SetAccessMode \x01<ETX>
[ INFO] [1627974881.155976273]: Sending  : <STX><STX><STX><STX><Len=0015>sWN EIHstCola 0x01 CRC:<0x09>
[ INFO] [1627974881.157470362]: Receiving: <STX>sWA EIHstCola <ETX>
[ INFO] [1627974881.358051068]: Sending  : <STX><STX><STX><STX><Len=0019>sRN FirmwareVersion CRC:<0x24>
[ INFO] [1627974881.359467937]: Receiving: <STX>sRA FirmwareVersion \x00\x09\x56\x31\x2e\x35\x30\x2e\x38\x20\x20<ETX>
[ INFO] [1627974881.560026676]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[ INFO] [1627974881.561496366]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[ INFO] [1627974881.762128653]: Sending  : <STX><STX><STX><STX><Len=0010>sRN ODoprh CRC:<0x41>
[ INFO] [1627974881.763592215]: Receiving: <STX>sRA ODoprh \x00\x00\x1f\xf0<ETX>
[ INFO] [1627974881.965969893]: Sending  : <STX><STX><STX><STX><Len=0010>sRN ODpwrc CRC:<0x52>
[ INFO] [1627974881.967356696]: Receiving: <STX>sRA ODpwrc \x00\x00\x08\xe3<ETX>
[ INFO] [1627974882.169685083]: Sending  : <STX><STX><STX><STX><Len=0016>sRN LocationName CRC:<0x55>
[ INFO] [1627974882.171206435]: Receiving: <STX>sRA LocationName \x00\x0b\x6e\x6f\x74\x20\x64\x65\x66\x69\x6e\x65\x64<ETX>
[ INFO] [1627974882.173358442]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMPoutputRange CRC:<0x5e>
[ INFO] [1627974882.174774508]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x09\xc4\xff\xff\x3c\xb0\x00\x1c\x3a\x90<ETX>
[ INFO] [1627974882.175004232]: Angle resolution of scanner is 0.25000 [deg]  (in 1/10000th deg: 0x9C4)
[ INFO] [1627974882.175095949]: [From:To] -5.00000 [deg] to 185.00000 [deg] (in 1/10000th deg: from 0xFFFF3CB0 to 0x1C3A90)
[ INFO] [1627974882.175206817]: MIN_ANG:   -1.658 [rad]  -95.000 [deg]
[ INFO] [1627974882.175299168]: MAX_ANG:    1.658 [rad]   95.000 [deg]
[ INFO] [1627974882.175531921]: Sending  : <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 0x00 0x09 0xc4 0xff 0xff 0x3c 0xb0 0x00 0x1c 0x3a 0x90 CRC:<0x8d>
[ INFO] [1627974882.176943390]: Receiving: <STX>sWA LMPoutputRange <ETX>
[ INFO] [1627974882.177252878]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMPoutputRange CRC:<0x5e>
[ INFO] [1627974882.178705405]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x09\xc4\xff\xff\x3c\xb0\x00\x1c\x3a\x90<ETX>
[ INFO] [1627974882.178973602]: Angle resolution of scanner is 0.25000 [deg]  (in 1/10000th deg: 0x9C4)
[... 68 lines cut ...]
[ INFO] [1627974882.234269672]: Safety fieldset 0, pointcounter = [  0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 ]
[ INFO] [1627974882.234562982]: Sending  : <STX><STX><STX><STX><Len=0032>sWN LMDscandatacfg 0x01 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x01 CRC:<0x42>
[ INFO] [1627974882.235910613]: Receiving: <STX>sWA LMDscandatacfg <ETX>
[ INFO] [1627974882.236163932]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMDscandatacfg CRC:<0x67>
[ INFO] [1627974882.237528123]: Receiving: <STX>sRA LMDscandatacfg \x01\x00\x01\x00\x00\x00\x00\x00\x00\x00\x01\x00\x01<ETX>
[ INFO] [1627974882.240912769]: sick_lms_5xx: activating field monitoring by lferec messages
[ INFO] [1627974882.242131511]: sick_lms_5xx: activating field monitoring by lidoutputstate messages
[ INFO] [1627974882.243330006]: sick_lms_5xx: activating field monitoring by lidinputstate messages
[ INFO] [1627974882.243606392]: Sending  : <STX><STX><STX><STX><Len=0016>sMN LMCstartmeas CRC:<0x68>
[ INFO] [1627974882.244963684]: Receiving: <STX>sAN LMCstartmeas \x00<ETX>
[ INFO] [1627974882.245256976]: Sending  : <STX><STX><STX><STX><Len=0007>sMN Run CRC:<0x19>
[ INFO] [1627974882.246628846]: Receiving: <STX>sAN Run \x01<ETX>
[ INFO] [1627974882.246927417]: Sending  : <STX><STX><STX><STX><Len=0017>sEN LMDscandata 0x01 CRC:<0x33>
[ INFO] [1627974882.248274974]: Receiving: <STX>sEA LMDscandata \x01<ETX>
[ INFO] [1627974882.248579655]: Sending  : <STX><STX><STX><STX><Len=0012>sEN LFErec 0x01 CRC:<0x42>
[ INFO] [1627974882.251079794]: Receiving: <STX>sEA LFErec \x01<ETX>
[ INFO] [1627974882.251409854]: Sending  : <STX><STX><STX><STX><Len=0020>sEN LIDoutputstate 0x01 CRC:<0x50>
[ INFO] [1627974882.252768984]: Receiving: <STX>sEA LIDoutputstate \x01<ETX>
[ INFO] [1627974882.253071745]: Sending  : <STX><STX><STX><STX><Len=0019>sEN LIDinputstate 0x01 CRC:<0x39>
[ INFO] [1627974882.254424628]: Receiving: <STX>sEA LIDinputstate \x01<ETX>
[ INFO] [1627974882.260705912]: SickScanServices: ros services initialized
free(): invalid pointer
[sick_lms_5xx-2] process has died [pid 73942, exit code -6, cmd /home/mitch/projects/masterarbeit/code/ros_ws/devel/lib/sick_scan/sick_generic_caller __name:=sick_lms_5xx __log:=/home/mitch/.ros/log/77bb77da-f42a-11eb-a365-8dc58b81f6cb/sick_lms_5xx-2.log].
log file: /home/mitch/.ros/log/77bb77da-f42a-11eb-a365-8dc58b81f6cb/sick_lms_5xx-2*.log
`

MitchFeike, 2021-08-03:
[ INFO] [1627974880.699658977]: sick_generic_caller V. 1.10.1
[ INFO] [1627974880.699709412]: Program arguments: /home/mitch/projects/masterarbeit/code/ros_ws/devel/lib/sick_scan/sick_generic_caller
[ INFO] [1627974880.699733828]: Program arguments: __name:=sick_lms_5xx
[ INFO] [1627974880.699742605]: Program arguments: __log:=/home/mitch/.ros/log/77bb77da-f42a-11eb-a365-8dc58b81f6cb/sick_lms_5xx-2.log
[ INFO] [1627974880.712020108]: Found sopas_protocol_type param overwriting default protocol:
[ INFO] [1627974880.712799863]: Binary protocol activated
[ INFO] [1627974880.712824622]: Start initialising scanner [Ip: 192.168.10.171] [Port: 2112]
[ INFO] [1627974880.737925913]: Publishing laserscan-pointcloud2 to cloud
[ INFO] [1627974880.747352103]: Parameter setting for <active_echo: 0>
[ INFO] [1627974880.949266824]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
[ INFO] [1627974880.955417806]: Receiving: <STX>sAN SetAccessMode \x01<ETX>
[ INFO] [1627974881.155976273]: Sending  : <STX><STX><STX><STX><Len=0015>sWN EIHstCola 0x01 CRC:<0x09>
[ INFO] [1627974881.157470362]: Receiving: <STX>sWA EIHstCola <ETX>
[ INFO] [1627974881.358051068]: Sending  : <STX><STX><STX><STX><Len=0019>sRN FirmwareVersion CRC:<0x24>
[ INFO] [1627974881.359467937]: Receiving: <STX>sRA FirmwareVersion \x00\x09\x56\x31\x2e\x35\x30\x2e\x38\x20\x20<ETX>
[ INFO] [1627974881.560026676]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[ INFO] [1627974881.561496366]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[ INFO] [1627974881.762128653]: Sending  : <STX><STX><STX><STX><Len=0010>sRN ODoprh CRC:<0x41>
[ INFO] [1627974881.763592215]: Receiving: <STX>sRA ODoprh \x00\x00\x1f\xf0<ETX>
[ INFO] [1627974881.965969893]: Sending  : <STX><STX><STX><STX><Len=0010>sRN ODpwrc CRC:<0x52>
[ INFO] [1627974881.967356696]: Receiving: <STX>sRA ODpwrc \x00\x00\x08\xe3<ETX>
[ INFO] [1627974882.169685083]: Sending  : <STX><STX><STX><STX><Len=0016>sRN LocationName CRC:<0x55>
[ INFO] [1627974882.171206435]: Receiving: <STX>sRA LocationName \x00\x0b\x6e\x6f\x74\x20\x64\x65\x66\x69\x6e\x65\x64<ETX>
[ INFO] [1627974882.173358442]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMPoutputRange CRC:<0x5e>
[ INFO] [1627974882.174774508]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x09\xc4\xff\xff\x3c\xb0\x00\x1c\x3a\x90<ETX>
[ INFO] [1627974882.175004232]: Angle resolution of scanner is 0.25000 [deg]  (in 1/10000th deg: 0x9C4)
[ INFO] [1627974882.175095949]: [From:To] -5.00000 [deg] to 185.00000 [deg] (in 1/10000th deg: from 0xFFFF3CB0 to 0x1C3A90)
[ INFO] [1627974882.175206817]: MIN_ANG:   -1.658 [rad]  -95.000 [deg]
[ INFO] [1627974882.175299168]: MAX_ANG:    1.658 [rad]   95.000 [deg]
[ INFO] [1627974882.175531921]: Sending  : <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 0x00 0x09 0xc4 0xff 0xff 0x3c 0xb0 0x00 0x1c 0x3a 0x90 CRC:<0x8d>
[ INFO] [1627974882.176943390]: Receiving: <STX>sWA LMPoutputRange <ETX>
[ INFO] [1627974882.177252878]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMPoutputRange CRC:<0x5e>
[ INFO] [1627974882.178705405]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x09\xc4\xff\xff\x3c\xb0\x00\x1c\x3a\x90<ETX>
[ INFO] [1627974882.178973602]: Angle resolution of scanner is 0.25000 [deg]  (in 1/10000th deg: 0x9C4)
[ INFO] [1627974882.182296865]: MIN_ANG (after command verification):   -1.658 [rad]  -95.000 [deg]
[ INFO] [1627974882.182436995]: MAX_ANG (after command verification):    1.658 [rad]   95.000 [deg]
[ INFO] [1627974882.182522244]: LMS 5xx detected overwriting output channel flag ID
[ INFO] [1627974882.182606482]: LMS 5xx detected overwriting resolution flag (only 8 bit supported)
[ INFO] [1627974882.182681610]: Reading safety fields
[ INFO] [1627974882.183022561]: Sending  : <STX><STX><STX><STX><Len=0012>sRN field000 CRC:<0x1d>
[ INFO] [1627974882.184500832]: Receiving: <STX>sRA field000 @\x00\x00\x00\x00\x00\x00\x00\x00\x00\x13\x88\xff\xff\x3c\xb0\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00<ETX>
[ INFO] [1627974882.184914595]: Sending  : <STX><STX><STX><STX><Len=0012>sRN field001 CRC:<0x1c>
[ INFO] [1627974882.186342974]: Receiving: <STX>sRA field001 @\x00\x00\x00\x00\x00\x00\x00\x00\x00\x13\x88\xff\xff\x3c\xb0\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00<ETX>
[ INFO] [1627974882.186674753]: Sending  : <STX><STX><STX><STX><Len=0012>sRN field002 CRC:<0x1f>
[ INFO] [1627974882.189197618]: Receiving: <STX>sRA field002 @\x00\x00\x00\x00\x00\x00\x00\x00\x00\x13\x88\xff\xff\x3c\xb0\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00<ETX>
[ INFO] [1627974882.189509089]: Sending  : <STX><STX><STX><STX><Len=0012>sRN field003 CRC:<0x1e>
[ INFO] [1627974882.192043816]: Receiving: <STX>sRA field003 @\x00\x00\x00\x00\x00\x00\x00\x00\x00\x13\x88\xff\xff\x3c\xb0\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00<ETX>
[ INFO] [1627974882.192353578]: Sending  : <STX><STX><STX><STX><Len=0012>sRN field004 CRC:<0x19>
[ INFO] [1627974882.194886897]: Receiving: <STX>sRA field004 @\x00\x00\x00\x00\x00\x00\x00\x00\x00\x13\x88\xff\xff\x3c\xb0\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00<ETX>
[ INFO] [1627974882.195224981]: Sending  : <STX><STX><STX><STX><Len=0012>sRN field005 CRC:<0x18>
[... cut ...]
[ INFO] [1627974882.254424628]: Receiving: <STX>sEA LIDinputstate \x01<ETX>
[ INFO] [1627974882.260705912]: SickScanServices: ros services initialized
free(): invalid pointer
[sick_lms_5xx-2] process has died [pid 73942, exit code -6, cmd /home/mitch/projects/masterarbeit/code/ros_ws/devel/lib/sick_scan/sick_generic_caller __name:=sick_lms_5xx __log:=/home/mitch/.ros/log/77bb77da-f42a-11eb-a365-8dc58b81f6cb/sick_lms_5xx-2.log].
log file: /home/mitch/.ros/log/77bb77da-f42a-11eb-a365-8dc58b81f6cb/sick_lms_5xx-2*.log

MitchFeike, 2021-08-03:
[ INFO] [1627974882.234269672]: Safety fieldset 0, pointcounter = [  0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 ]
[ INFO] [1627974882.234562982]: Sending  : <STX><STX><STX><STX><Len=0032>sWN LMDscandatacfg 0x01 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x01 CRC:<0x42>
[ INFO] [1627974882.235910613]: Receiving: <STX>sWA LMDscandatacfg <ETX>
[ INFO] [1627974882.236163932]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMDscandatacfg CRC:<0x67>
[ INFO] [1627974882.237528123]: Receiving: <STX>sRA LMDscandatacfg \x01\x00\x01\x00\x00\x00\x00\x00\x00\x00\x01\x00\x01<ETX>
[ INFO] [1627974882.240912769]: sick_lms_5xx: activating field monitoring by lferec messages
[ INFO] [1627974882.242131511]: sick_lms_5xx: activating field monitoring by lidoutputstate messages
[ INFO] [1627974882.243330006]: sick_lms_5xx: activating field monitoring by lidinputstate messages
[ INFO] [1627974882.243606392]: Sending  : <STX><STX><STX><STX><Len=0016>sMN LMCstartmeas CRC:<0x68>
[ INFO] [1627974882.244963684]: Receiving: <STX>sAN LMCstartmeas \x00<ETX>
[ INFO] [1627974882.245256976]: Sending  : <STX><STX><STX><STX><Len=0007>sMN Run CRC:<0x19>
[ INFO] [1627974882.246628846]: Receiving: <STX>sAN Run \x01<ETX>
[ INFO] [1627974882.246927417]: Sending  : <STX><STX><STX><STX><Len=0017>sEN LMDscandata 0x01 CRC:<0x33>
[ INFO] [1627974882.248274974]: Receiving: <STX>sEA LMDscandata \x01<ETX>
[ INFO] [1627974882.248579655]: Sending  : <STX><STX><STX><STX><Len=0012>sEN LFErec 0x01 CRC:<0x42>
[ INFO] [1627974882.251079794]: Receiving: <STX>sEA LFErec \x01<ETX>
[ INFO] [1627974882.251409854]: Sending  : <STX><STX><STX><STX><Len=0020>sEN LIDoutputstate 0x01 CRC:<0x50>
[ INFO] [1627974882.252768984]: Receiving: <STX>sEA LIDoutputstate \x01<ETX>
[ INFO] [1627974882.253071745]: Sending  : <STX><STX><STX><STX><Len=0019>sEN LIDinputstate 0x01 CRC:<0x39>
[ INFO] [1627974882.254424628]: Receiving: <STX>sEA LIDinputstate \x01<ETX>
[ INFO] [1627974882.260705912]: SickScanServices: ros services initialized
free(): invalid pointer
[sick_lms_5xx-2] process has died [pid 73942, exit code -6, cmd /home/mitch/projects/masterarbeit/code/ros_ws/devel/lib/sick_scan/sick_generic_caller __name:=sick_lms_5xx __log:=/home/mitch/.ros/log/77bb77da-f42a-11eb-a365-8dc58b81f6cb/sick_lms_5xx-2.log].
log file: /home/mitch/.ros/log/77bb77da-f42a-11eb-a365-8dc58b81f6cb/sick_lms_5xx-2*.log

michael1309, 2021-08-03:
Hi,

1. could you try to send us a callstack/backtrace by using the gdb as mentioned in ticket https://github.com/SICKAG/sick_scan/issues/85
? 
2. could you try to start the LMS511 by using the default launch file and report, whether the node is crashing?
3. What hardware do you use?
Thanks in advance

MitchFeike, 2021-08-04:
Hello,

thank you for your answer.
I checked the ticket #85 and the FW (https://github.com/SICKAG/sick_scan/issues/85#issuecomment-624042085) of my sensor is very old too.
A couple of days ago I tried to update in SOPAS but I failed cause I wasn't able to push the "update sensor" button. 
How do update the sensor?
Do I need other login data?

michael1309, 2021-08-04:
Hi,
I recommend to contact the SICK support to upgrade your firmware using "SOPAS ET". You can upload with the correct user rights (e.g. "service") to a more recent firmware version by uploading a so called SPK-package containing the firmware.

MitchFeike, 2021-08-04:
Hi, 
thank you very much for your help.
I'll try to update the firmware and try the ROS driver with the default launch file in the next couple of hours. 
I'll let you know what I achieved.

michael1309, 2021-08-04:
Hi,

can you try to use the "gdb" as described in the ticket #85 ?

MitchFeike, 2021-08-04:
Hi,

after firmware update and setting back to default settings it works for me. 
Thank you very much for your help.

MitchFeike, 2021-08-04:
The Sopas Engineering Tool tells me that the min. angle is -5° and the max. angle is 185° but if I set this in the launch file there are occurs an error. 

```
[ INFO] [1628080174.954590170]: Angle resolution of scanner is 1.00000 [deg]  (in 1/10000th deg: 0x2710)
[ INFO] [1628080174.954658591]: [From:To] -5.00000 [deg] to 185.00000 [deg] (in 1/10000th deg: from 0xFFFF3CB0 to 0x1C3A90)
[ INFO] [1628080174.954707295]: MIN_ANG:   -0.087 [rad]   -5.000 [deg]
[ INFO] [1628080174.954802502]: MAX_ANG:    3.229 [rad]  185.000 [deg]
[ INFO] [1628080174.954998441]: Sending  : <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 0x00 0x27 0x10 0x00 0x0c 0xf8 0x50 0x00 0x29 0xf6 0x30 CRC:<0x06>
[ INFO] [1628080174.956307965]: Receiving: <STX>sFA\x00\x04<ETX>
[ERROR] [1628080174.956474962]: Error Sopas answer mismatch Error unexpected Sopas Answer for request <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 0x00 0x27 0x10 0x00 0x0c 0xf8 0x50 0x00 0x29 0xf6 0x30 CRC:<0x06>Answer= >>>sFA\x00\x04<<<

[ INFO] [1628080174.956903919]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMPoutputRange CRC:<0x5e>
[ INFO] [1628080174.958274907]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x27\x10\xff\xff\x3c\xb0\x00\x1c\x3a\x90<ETX>
[ INFO] [1628080174.958494885]: Angle resolution of scanner is 1.00000 [deg]  (in 1/10000th deg: 0x2710)
[ INFO] [1628080174.961523812]: MIN_ANG (after command verification):   -1.658 [rad]  -95.000 [deg]
[ INFO] [1628080174.961620638]: MAX_ANG (after command verification):    1.658 [rad]   95.000 [deg]

```

michael1309, 2021-08-04:
Please note that most SICK lidars specify the X-axis to the right. However, the ROS driver consistently uses the ROS convention with the X-axis to the front. This explains the difference. Therefore, please use the range from -95 degrees to +95 degrees in rad (cf. launch file for the LMS5xx).

michael1309, 2021-08-04:
BTW: Could you mention the differences between the default launch file and your changes that caused problems? Then we could look through this again in the future.

michael1309, 2021-08-18:
Close due to inactivity. We guess are comments solved and clarified this issue.

closed 2021-08-18 by michael1309