SICKAG/sick_scan #137: Localization stops in Lidar-loc software when I publish lidar data to ros system using sick_scan package

Dheenu-kasinathan, 2021-08-03:
Hi,

I am currently working on 2D LiDAR sensors LMS1000 along with lidar loc for localization. whenever i ran the sick_scan package to publish the lidar data to ros system, the lidar-loc stops localizing. I believe it has something to do with the port configuration. Is there any way I can configure the lidar to broadcast the data in different port? Or can i get the lidar data from the sim_lidar_loc ros package?  Let me know if I am missing something. Any help would be appreciated

sim_loc_driver.yaml
```
sick_lidar_localization:

  # Driver configuration. See Operation-Instruction-v1.1.0.241R.pdf, page 51, "IP port number and protocol" for default tcp ports.
  driver:
    localization_controller_default_ip_address: "192.168.0.1"                 
    result_telegrams_tcp_port: 2201                                            
    cola_telegrams_tcp_port:   2111                                            
    cola_binary: 0                                                             
    tcp_connection_retry_delay: 1.0                                            
    result_telegrams_topic: "/sick_lidar_localization/driver/result_telegrams" 
    result_telegrams_frame_id: "sick_lidar_localization"                      
    diagnostic_topic: "/sick_lidar_localization/driver/diagnostic"             
    diagnostic_frame_id: "sick_lidar_localization"                             
    monitoring_rate: 1.0                                                       
    monitoring_message_timeout: 1.0                                            
    point_cloud_topic: "/2dcloud"                                                
    point_cloud_frame_id: "map"                 
    tf_parent_frame_id: "map"                                         
    tf_child_frame_id: "base_link"                          

  # Configuration for time sync service
  time_sync:
    cola_response_timeout:      1.0  
    software_pll_fifo_length:   7    
    time_sync_rate:             0.1    
    time_sync_initial_rate:     1.0 
    time_sync_initial_length:  10
```
sick_lms_1xxx.launch
```
<?xml version="1.0"?>
<!-- FOR FUTURE USE. NOT SUPPORTED NOW. DO NOT USE IT. -->
<launch>
    <arg name="hostname" default="192.168.0.10"/>
    <!-- robot_description and robot_state_publisher hier evtl. einbauen  -->
    <node name="sick_lms_1xxx" pkg="sick_scan" type="sick_generic_caller" respawn="false" output="screen">
        <!-- default values: -->
        <!--
          <param name="scanner_type" type="string" value="sick_lms_1xxx"
          <param name="min_ang" type="double" value="-2.35619449019" />
          <param name="max_ang" type="double" value="2.35619449019" />
          <param name="intensity" type="bool" value="True" />
          <param name="skip" type="int" value="0" />
          <param name="frame_id" type="str" value="laser" />
          <param name="time_offset" type="double" value="-0.001" />
          <param name="publish_datagram" type="bool" value="False" />
          <param name="subscribe_datagram" type="bool" value="false" />
          <param name="device_number" type="int" value="0" />
          <param name="range_min" type="double" value="0.05" />
        -->
        <param name="scanner_type" type="string" value="sick_lms_1xxx"/>
        <param name="range_max" type="double" value="25.0"/>
        <param name="hostname" type="string" value="$(arg hostname)"/>
        <param name="port" type="string" value="2111"/>
        <param name="timelimit" type="int" value="6"/>
        <param name="intensity" type="bool" value="true"/>
        <param name="intensity_resolution_16bit" type="bool" value="false"/>
        <param name="min_ang" type="double" value="-2.391"/>
        <param name="max_ang" type="double" value="2.391"/>

    </node>
</launch>

```

Thanks

rostest, 2021-08-04:
Thanks for your input. The combination of sick_lidar_localization plus sick_scan is currently not supported. Despite of port configuration issues, conflicts in concurrent setups and configurations may cause undesired behaviour. I.e. you have to choose between runing localization software or using sick_scan for processing the raw data of the lidar. We recommend the usage of SICK lidar localization for localization and navigation.  You can use the Localization Result telegrams to retrieve further information about the current lidar pose.

Dheenu-kasinathan, 2021-08-06:
Thanks for your response. I wanted to use the data from lidar (lms100) as an extra layer for my obstacle detection. I noticed that the sick_lidar_loc ros package is outputting the 2d pointcloud. I still need to validate the sick_lidar_loc output cloud is same as the output of lms1000.

michael1309, 2021-08-18:
Can we close this ticket? As mentioned by us your request is a little bit beyond the scope of this driver.

rostest, 2021-09-20:
Clarified.

closed 2021-09-20 by rostest