SICKAG/sick_scan #140: When i try to connect MRS 1000 with Linux system, connection is not establishing and it throws the error

LOKAKSHAI, 2021-08-18:
1629282227.8051 ERROR: Tcp::open: Failed to open TCP connection to 192.168.255.22, aborting.

[ INFO] [1629282227.802324073]: Publishing laserscan-pointcloud2 to cloud
1629282227.8051 ERROR: Tcp::open: Failed to open TCP connection to 192.168.255.22, aborting.
[ INFO] [1629282227.807395482]: Parameter setting for <active_echo: 0>
[ INFO] [1629282228.008993871]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
1629282228.0092 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1629282233.869723263]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ INFO] [1629282233.869861733]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1629282233.870166166]: Receiving: <STX><ETX>
[ INFO] [1629282233.870242166]: SOPAS Communication -Error unexpected Sopas Answer for request <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>

[ INFO] [1629282233.870516308]: Sending  : <STX>sMN SetAccessMode 3 F4724744<ETX>
1629282233.8706 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1629282239.707449705]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ INFO] [1629282239.707626225]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1629282239.708003850]: Receiving: <STX><ETX>
[ INFO] [1629282239.708116647]: SOPAS Communication -Error unexpected Sopas Answer for request <STX>sMN SetAccessMode 3 F4724744<ETX>

[ERROR] [1629282239.708382807]: Error Access Mode
[ INFO] [1629282239.909224756]: Sending  : <STX>sWN EIHstCola 1<ETX>
1629282239.9094 Warning: Tcp::write: Failed to send data to socket.
[ERROR] [1629282245.881525859]: no answer received after 5000 ms. Maybe sopas mode is wrong.

michael1309, 2021-08-18:
OK. Here are some hints for you.

Please set this Ethernet adapter of your PC to a static IP address in the address space 192.168.0.xxx (we recommend 192.168.0.100).
Then please try to ping the Lidar at the address 192.168.0.1.
If this is successful, also use netcat to check if the Lidar can be reached on port 2112 by running the following command:

'nc -z -v -w5 192.168.0.1 2112'

For more information, see e.g.

            https://github.com/SICKAG/sick_scan/blob/master/doc/ipconfig/ipconfig.md
For setting up a static IP, you can follow the standard documentation for Ubuntu, etc.

michael1309, 2021-08-18:
BTW: It is a little bit unusual to use the address space 192.168.255.xxx. We recommend to use a more common subnet like 192.168.0.xxx.

rostest, 2021-09-20:
Closed due to inactivity.

closed 2021-09-20 by rostest