SICKAG/sick_scan #141: SICK LMS5**使用过程中办错

zhshwind, 2021-08-28:
SICK LMS5** 使用sick_scan进行测试的过程中出现了以下的报错；

在同时连接两个激光雷达的时候报错更加频繁，请问这种问题如何解决？
[image]
[image]
[image]
[image]
[image]
[image]

zhshwind, 2021-09-10:
> Hello ,
>
> Unfortunately I was not able to reproduce the error.
> You use the field evaluation or? How is it configured ?
> Does the error occur immediately after the start (so it looks in the logs)?
>
> Which variant of the software do you use? Try the latest version from this repo.

Hello：
    The problem we are encountering at present is: every time after power on, the lidar ros node will report an error.
    Most of the time the error is TCP Time_out. This kind of error, the lidar data can be read normally.
    Another type of error-reporting lidar cannot be used normally. The error message is shown in the figure below.

    We don't know how this problem arose and how it needs to be solved. Hope to get your professional guidance. 
    Thanks!

    

|
郑爽
***@***.***
|
签名由网易邮箱大师定制

rostest, 2021-09-10:
Hello,

thanks for your report. I can't see the figure, but it looks like the lidar is starting in COLA-Ascii mode after cold start, while the driver is expecting the COLA-Binary mode. In this case, a timeout can occure due to protocol switching.

Using COLA-Binary communication is recommended. Please make sure to set the lidar permanently to COLA-Binary. You can do this by using SOPAS ET after login to "authorized client" level. After switching you should save the parameter to the internal memory of the scanner (EEPROM). After a new cold start, check in SOPAS ET that the setting is actually stored in the lidar .

You can find additional information in the closed issues #130 and #6.

rostest, 2021-09-20:
Closed due to inactivity.

closed 2021-09-20 by rostest

zhshwind, 2021-09-26:
Hello：
    Thank you for your response.

    The problem we are currently encountering should not be the reason you mentioned.
    I checked the current settings, the current lidar permanently is COLA-Binary. 
|
郑爽
***@***.***
|
签名由网易邮箱大师定制

Closed #141.