SICKAG/sick_scan #143: Diagnostics Message has level 2 status

5730289021-NN, 2021-09-17:
I subscribed to /diagnostics and I got these message:

```
header: 
  seq: 471
  stamp: 
    secs: 1631858068
    nsecs: 641215806
  frame_id: ''
status: 
  - 
    level: 2
    name: "sick_tim_7xx: /scan topic status"
    message: "No events recorded.; No data since last update."
    hardware_id: "none"
    values: 
      - 
        key: "Events in window"
        value: "0"
      - 
        key: "Events since startup"
        value: "0"
      - 
        key: "Duration of window (s)"
        value: "10.533250"
      - 
        key: "Actual frequency (Hz)"
        value: "0.000000"
      - 
        key: "Target frequency (Hz)"
        value: "15.000000"
      - 
        key: "Minimum acceptable frequency (Hz)"
        value: "13.500000"
      - 
        key: "Maximum acceptable frequency (Hz)"
        value: "16.500000"
      - 
        key: "Earliest timestamp delay"
        value: "No data"
      - 
        key: "Latest timestamp delay"
        value: "No data"
      - 
        key: "Earliest acceptable timestamp delay"
        value: "-1.000000"
      - 
        key: "Latest acceptable timestamp delay"
        value: "0.086667"
      - 
        key: "Late diagnostic update count"
        value: "0"
      - 
        key: "Early diagnostic update count"
        value: "0"
      - 
        key: "Zero seen diagnostic update count"
        value: "0"
```

I've checked on [DiagnostocStatus](http://docs.ros.org/en/api/diagnostic_msgs/html/msg/DiagnosticStatus.html), level 2 indicates an error. I'm wondering what's the error here

The launch file node configuration is
```
    <node name="sick_tim_7xx" pkg="sick_scan" type="sick_generic_caller" respawn="false" output="screen">
        <!--<param name="robot_description" command="$(find xacro)/xacro.py '$(find sick_scan)/urdf/example.urdf.xacro'" />-->
        <param name="scanner_type" type="string" value="sick_tim_7xxS"/>
        <!-- <param name="range_min" type="double" value="0.2"/> -->

        <param name="min_ang" type="double" value="-1.5"/>
        <param name="max_ang" type="double" value="1.5"/>
        <param name="use_binary_protocol" type="bool" value="True"/>
        <param name="range_max" type="double" value="30.0"/>
        <param name="intensity" type="bool" value="True"/>
        <param name="hostname" type="string" value="192.168.1.24"/>
        <param name="cloud_topic" type="string" value="cloud"/>
        <param name="frame_id" type="str" value="cloud"/>
        <param name="port" type="string" value="2112"/>
        <param name="timelimit" type="int" value="5"/>
        <param name="start_services" type="bool" value="True"/> <!-- start ros service for cola commands -->
        <param name="activate_lferec" type="bool" value="True"/> <!-- activate field monitoring by lferec messages -->
        <param name="activate_lidoutputstate" type="bool" value="True"/> <!-- activate field monitoring by lidoutputstate messages -->
        <param name="activate_lidinputstate" type="bool" value="True"/> <!-- activate field monitoring by lidinputstate messages -->
    </node>
```

I'm not sure whether this is normal or not. Please let me know if something is truly error.

5730289021-NN, 2021-09-17:
Operating System: Ubuntu 20.04
ROS Noetic
Installed via ```sudo apt install ros-noetic-sick-scan```

rostest, 2021-09-17:
Thanks for your report. sick_scan publishes diagnostic messages only in case of errors. The messages you see are "no events / no data" messages and do not indicate an error. These messages can safely be ignored.
Nevertheless, a diagnostic level 2 can be misleading. We will therefore have a closer look to further status messages. Currently, the messages are ment for diagnosis in case of errors and can be ignored as long as the scanner works as expected.

rostest, 2021-09-20:
Until end of year 2021 we will update the repository sick_scan and sick_scan2 to a combined version. By generating this update we will also update the diagnostic message handling by following your hint. We will publish diagnostic default status message with diagnostic level 0 (or with level 2 in case of an error). So we would like to postpone this issue and close it.

closed 2021-09-20 by rostest