SICKAG/sick_scan #151: Inconsistent time_increment, scan_time and angle_increment

ZSeymour1, 2021-11-10:
Hi,

I apologise if this has been covered somewhere and I've missed it when searching, but I receive the below warning message when running the sick_tim_5xx.launch file. The data still comes through (in both SOPAS ET and on RVIZ), so I'm not sure how serious this issue is. Regardless, it would be nice to know how to set the correct parameters (which I was unable to find) to avoid this. Essentially I don't know how to access and set this "time_increment" parameter (not in the launch file or on rosparam list).

"The time_increment, scan_time and angle_increment values reported by the scanner are inconsistent! Expected time_increment: 0.000185185, reported time_increment: 0.000061728. Perhaps you should set the parameter time_increment to the expected value."

For reference, I'm using ROS Noetic with Ubuntu 20.04, I've installed from a binary (sudo apt-get install ros-noetic-sick-scan), since I was having trouble with the source installation.

Many thanks for any help!

michael1309, 2021-11-10:
@ZSeymour1 
Unfortunately, We could not reproduce this behavior with a TIM 561. Which scanner do you use exactly?
What is the update rate of your scanner ? 
you can read it with this command:
```
rostopic hz /cloud
```
Can you also post your entire launchfile so we can figure out the reason better ?

ZSeymour1, 2021-11-11:
Hi,

We're using the SICK TIM 571 (P/N: 1075091) with the 25m range (outdoor version). The update rate (using rostopic hz /cloud) is approximately 15Hz (see below for copy-pasted data). 

I've also attached the launch file (but as a .txt not .launch), which is the default one provided in the package, with the default hostname IP changed to "190.168.1.71".

_average rate: 15.007
	min: 0.063s max: 0.070s std dev: 0.00130s window: 15
average rate: 15.007
	min: 0.063s max: 0.070s std dev: 0.00095s window: 30
average rate: 15.001
	min: 0.063s max: 0.070s std dev: 0.00082s window: 45
average rate: 15.002
	min: 0.063s max: 0.070s std dev: 0.00075s window: 60
average rate: 15.000
	min: 0.063s max: 0.070s std dev: 0.00070s window: 75
average rate: 15.001
	min: 0.063s max: 0.070s std dev: 0.00067s window: 90
average rate: 15.000
	min: 0.063s max: 0.070s std dev: 0.00066s window: 92_

[sick_tim_5xx.txt](https://github.com/SICKAG/sick_scan/files/7519734/sick_tim_5xx.txt)

michael1309, 2022-01-18:
We have tested with the TiM571, using both sick_scan and sick_scan_xd on ROS-1 Noetic. Unfortunately, the error is not reproducable in our system. There are no error messages and the frequencies are as expected (expected and reported time increment at 6.172e-05). As long as TiM571 and driver run as expected and provide the scanner data, it's save to ignore the warning.

closed 2022-01-18 by michael1309