SICKAG/sick_scan #155: Need to restart NAV310 two times until proper scan is provided

HannesBachter, 2022-01-04:
I am using a Nav310 Scanner and set the communication Protocol to `CoLa Binary`:

[image]
---
When I start the scanner first after shutdown, the output from the launch complains about a erroneous setting (which should be set properly as depicted above):
```
SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.13
 * /sick_nav_3xx/cloud_topic: cloud
 * /sick_nav_3xx/frame_id: base_link_laser
 * /sick_nav_3xx/hostname: 192.168.1.11
 * /sick_nav_3xx/intensity: True
 * /sick_nav_3xx/max_ang: 3.14159265359
 * /sick_nav_3xx/min_ang: -3.14159265359
 * /sick_nav_3xx/port: 2112
 * /sick_nav_3xx/scanner_type: sick_nav_3xx
 * /sick_nav_3xx/timelimit: 5
 * /sick_nav_3xx/use_binary_protocol: True

NODES
  /
    sick_nav_3xx (sick_scan/sick_generic_caller)

ROS_MASTER_URI=http://192.168.1.203:11311

process[sick_nav_3xx-1]: started with pid [7020]
[ INFO] [1640021298.804037135]: ${node}(main): sick_generic_caller V. 1.8.0
[ INFO] [1640021298.804075041]: ${node}(main): Program arguments: /home/robot/git/lts_test_ws/devel/lib/sick_scan/sick_generic_caller
[ INFO] [1640021298.804086890]: ${node}(main): Program arguments: __name:=sick_nav_3xx
[ INFO] [1640021298.804096872]: ${node}(main): Program arguments: __log:=/home/robot/.ros/log/022c2f60-61b1-11ec-b986-8bf27e0056b3/sick_nav_3xx-1.log
[ INFO] [1640021298.814802218]: /sick_nav_3xx(string)): Found sopas_protocol_type param overwriting default protocol:
[ INFO] [1640021298.815534528]: /sick_nav_3xx(string)): Binary protocol activated
[ INFO] [1640021298.815554942]: /sick_nav_3xx(string)): Start initialising scanner [Ip: 192.168.1.11] [Port: 2112]
[ INFO] [1640021298.834744275]: /sick_nav_3xx(SickScanCommon::SickScanCommon): Publishing laserscan-pointcloud2 to cloud
[ INFO] [1640021298.838649461]: /sick_nav_3xx(SickScanCommon::init_scanner): Parameter setting for <active_echo: 0>
[ INFO] [1640021299.040025108]: /sick_nav_3xx(SickScanCommon::sendSopasAndCheckAnswer): Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
[ERROR] [1640021304.982609759]: /sick_nav_3xx(SickScanCommonTcp::readWithTimeout): no answer received after 5000 ms. Maybe sopas mode is wrong.

[ INFO] [1640021304.982858587]: /sick_nav_3xx(SickScanCommonTcp::sendSOPASCommand): sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1640021304.983290159]: /sick_nav_3xx(SickScanCommon::sendSopasAndCheckAnswer): Receiving: <STX><ETX>
[ INFO] [1640021304.983460338]: /sick_nav_3xx(SickScanCommon::sendSopasAndCheckAnswer): SOPAS Communication -Error unexpected Sopas Answer for request <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>

[ INFO] [1640021304.984001386]: /sick_nav_3xx(SickScanCommon::sendSopasAndCheckAnswer): Sending  : <STX>sMN SetAccessMode 3 F4724744<ETX>
[ INFO] [1640021304.986688647]: /sick_nav_3xx(SickScanCommon::sendSopasAndCheckAnswer): Receiving: <STX>sAN SetAccessMode 1<ETX>
[ INFO] [1640021305.187408409]: /sick_nav_3xx(SickScanCommon::sendSopasAndCheckAnswer): Sending  : <STX>sWN EIHstCola 1<ETX>
[ INFO] [1640021305.189152660]: /sick_nav_3xx(SickScanCommon::sendSopasAndCheckAnswer): Receiving: <STX>sWA EIHstCola<ETX>
[ INFO] [1640021305.189732172]: /sick_nav_3xx(SickScanCommon::init): Failed to init scanner Error Code: 1
Waiting for timeout...
If the communication mode set in the scanner memory is different from that used by the driver, the scanner's communication mode is changed.
This requires a restart of the TCP-IP connection, which can extend the start time by up to 30 seconds. There are two ways to prevent this:
1. [Recommended] Set the communication mode with the SOPAS ET software to binary and save this setting in the scanner's EEPROM.
2. Use the parameter "use_binary_protocol" to overwrite the default settings of the driver.
[ INFO] [1640021305.192466270]: /sick_nav_3xx(string)): Start initialising scanner [Ip: 192.168.1.11] [Port: 2112]
[ WARN] [1640021305.192660107]: /sick_nav_3xx(SickScanCommonTcp::close_device): Disconnecting TCP-Connection.
1640021306.1899 Info: Tcp::readInputData: Read 0 bytes - connection is lost!
sick_scan driver exiting.
[ INFO] [1640021306.248162506]: /sick_nav_3xx(SickScanCommon::SickScanCommon): Publishing laserscan-pointcloud2 to cloud
[ INFO] [1640021306.252102786]: /sick_nav_3xx(SickScanCommon::init_scanner): Parameter setting for <active_echo: 0>
[ INFO] [1640021306.453086720]: /sick_nav_3xx(SickScanCommon::sendSopasAndCheckAnswer): Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
1640021306.4552 ERROR: Tcp::readInputData: Failed to read data from socket, aborting!
[ERROR] [1640021312.413630547]: /sick_nav_3xx(SickScanCommonTcp::readWithTimeout): no answer received after 5000 ms. Maybe sopas mode is wrong.

[ INFO] [1640021312.413844711]: /sick_nav_3xx(SickScanCommonTcp::sendSOPASCommand): sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1640021312.414322652]: /sick_nav_3xx(SickScanCommon::sendSopasAndCheckAnswer): Receiving: <STX><ETX>
[ INFO] [1640021312.414572331]: /sick_nav_3xx(SickScanCommon::sendSopasAndCheckAnswer): SOPAS Communication -Error unexpected Sopas Answer for request <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>

[... 253 lines cut ...]
```

The scan angles seem more trustworthy (although they are switched as reported in https://github.com/SICKAG/sick_scan/issues/110) and the frequency updated to `20 Hz` (instead of `8 Hz` as above): 
```
rostopic echo /scan -n 1
header: 
  seq: 0
  stamp: 
    secs: 1640021595
    nsecs: 728102947
  frame_id: "base_link_laser"
angle_min: 3.1415927410125732
angle_max: -3.128502607345581
angle_increment: -0.013089969754219055
time_increment: 0.00010416666918899864
scan_time: 0.05000000074505806
range_min: 0.05000000074505806
range_max: 100.0
ranges: [6.824000358581543, 6.836000442504883, 6.884000301361084, 6.87600040435791, 6.880000114440918, 6.908000469207764, 6.892000198364258, 6.900000095367432, 6.924000263214111, 6.936000347137451, 6.948000431060791, 6.960000514984131, 7.348000526428223, 7.308000564575195, 7.012000560760498, 7.484000205993652, 7.492000579833984, 7.520000457763672, 7.544000148773193, 7.592000484466553, 7.5960001945495605, 7.6080002784729, 7.660000324249268, 7.616000175476074, 8.260000228881836, 8.332000732421875, 8.472000122070312, 8.368000030517578, 8.0, 7.680000305175781, 7.448000431060791, 7.432000160217285, 7.504000186920166, 7.556000232696533, 29.716001510620117, 43.220001220703125, 35.816001892089844, 28.720001220703125, 0.0, 44.540000915527344, 18.088001251220703, 18.172000885009766, 0.0, 0.0, 46.52000045776367, 12.324000358581543, 31.2080020904541, 17.772001266479492, 48.320003509521484, 48.80000305175781, 29.016000747680664, 29.056001663208008, 37.7400016784668, 22.076000213623047, 14.092000961303711, 14.24000072479248, 13.144001007080078, 35.16400146484375, 21.908000946044922, 19.064001083374023, 3.880000114440918, 3.876000165939331, 3.8680002689361572, 4.076000213623047, 3.932000160217285, 3.9680001735687256, 4.0360002517700195, 4.048000335693359, 4.084000110626221, 4.028000354766846, 3.9840002059936523, 4.384000301361084, 4.496000289916992, 4.576000213623047, 4.668000221252441, 4.784000396728516, 4.880000114440918, 4.996000289916992, 5.116000175476074, 5.236000061035156, 5.368000030517578, 5.504000186920166, 5.648000240325928, 5.812000274658203, 5.956000328063965, 6.140000343322754, 6.308000087738037, 6.52400016784668, 6.744000434875488, 6.956000328063965, 7.204000473022461, 7.484000205993652, 7.760000228881836, 8.079999923706055, 8.40000057220459, 8.756000518798828, 8.688000679016113, 8.684000015258789, 10.204000473022461, 10.796000480651855, 11.476000785827637, 8.92400074005127, 8.616000175476074, 8.596000671386719, 8.576000213623047, 8.54800033569336, 8.528000831604004, 8.516000747680664, 8.480000495910645, 8.484000205993652, 8.476000785827637, 8.472000122070312, 8.456000328063965, 8.444000244140625, 8.43600082397461, 8.444000244140625, 8.43600082397461, 8.432000160217285, 8.444000244140625, 8.43600082397461, 8.440000534057617, 8.43600082397461, 8.452000617980957, 8.460000038146973, 8.460000038146973, 8.476000785827637, 8.476000785827637, 8.504000663757324, 8.51200008392334, 8.520000457763672, 8.539999961853027, 8.576000213623047, 8.576000213623047, 8.600000381469727, 8.631999969482422, 8.660000801086426, 8.688000679016113, 8.704000473022461, 8.744000434875488, 8.76400089263916, 8. [... cut ...] 0, 8896.0, 7232.0, 7168.0, 7360.0, 9856.0, 7168.0, 7040.0, 7168.0, 7808.0, 6976.0, 6848.0, 0.0, 7040.0, 3264.0, 3264.0, 6400.0, 5696.0, 4480.0, 5312.0, 3520.0, 7232.0, 2496.0, 0.0, 0.0, 6272.0, 6208.0, 5888.0, 2176.0, 0.0, 2624.0, 2688.0, 8704.0, 3584.0, 3840.0, 3840.0, 3840.0, 3904.0, 4224.0, 4416.0, 7936.0, 11392.0, 10688.0, 4416.0, 6080.0, 7680.0, 7744.0, 7680.0, 7744.0, 7872.0, 8448.0, 12288.0, 15680.0, 16448.0, 13952.0, 10048.0]
---

The scan represents the environment properly:
```
[image]

michael1309, 2022-01-05:
Within SOPAS ET please set the communication to SOPAS Binary and store this value into the EEPROM of the lidar, so that the lidar starts directly with SOPAS Binary communication after cold start. 

You can recognize a correct setting to SOPAS Binary by the fact that on the first command
```
Sending : <STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
```
is answered as follows:
```
Receiving: <STX>sAN SetAccessMode \x01<ETX>
```

If you have made these settings, we recommend that you start the Lidar with cold start. We have seen the behavior you describe (scans not being acquired correctly) when switching modes without cold start in our testing as well. The best way to solve this problem was to set the lidar to SOPAS Binary by storing it into the EEPROM (see description above) and then cold start the lidar correctly.

HannesBachter, 2022-01-10:
We could fix the issue by permanently storing the communication protocol to binary

closed 2022-01-10 by HannesBachter