SICKAG/sick_scan #158: Hello, I found that the ROS driver will crash during use. What is the reason?

WangQJuan, 2022-02-21:
Hello, I found that the ROS driver will crash during use. What is the reason
[image]
?

rostest, 2022-02-21:
Thanks for reporting! The error seems to have occurred before ("Error acitivating LIDouputstate message"), causing sick_generic_caller to terminate. Please post the complete output from start, or at least the complete error message.

WangQJuan, 2022-02-21:
Hello, thank you for your support. The following is the error information
[image]
[image]
[image]
.

rostest, 2022-02-21:
Thanks! I'll have a closer look to the error and get back on this as soon as possible. If possible, consider an update to the sick_scan_xd driver on https://github.com/SICKAG/sick_scan_xd with the most recent version.

WangQJuan, 2022-02-21:
Thank you for your support and wait for your good news.

rostest, 2022-02-28:
LFErec, LIDinputstate and LIDoutputstate have been activated after switching to measurement mode. Depending on timing, this could lead to mismatching scandata and sopas response messages. The bug is fixed in the latest release 1.12.0 by changing the sopas startup sequence.
Please update the sources and rebuild the driver. Thanks again!

gowtham212, 2022-05-05:
Hi while I am launching this file sick_scan sick_lms_1xx.launch 
roslaunch sick_scan sick_lms_1xx.launch 
I get code from devel branch https://github.com/SICKAG/sick_scan.git
I got error like this

[image]

rostest, 2022-05-06:
Thanks for reporting. Please use the latest release https://github.com/SICKAG/sick_scan/releases/tag/1.12.1 or the master branch. Checkout using `git clone https://github.com/SICKAG/sick_scan.git`
We recommend to switch to https://github.com/SICKAG/sick_scan_xd for the most recent driver version.

hasanzfdn, 2022-07-17:
ı got same error, tried 1.12.1 release and master branch but still doesn't solved.

rostest, 2022-07-18:
Thanks for reporting. Please use the repository on https://github.com/SICKAG/sick_scan_xd with the most recent releases.

hasanzfdn, 2022-07-18:
Thank you for replying. I solved by close the wi-fi and unplug-plug the ethernet cable.

closed 2022-07-19 by rostest