SICKAG/sick_scan #159: Laser data timing problem

WangQJuan, 2022-02-21:
Have you ever found the problem that two frames of laser data will be separated by a few milliseconds？

rostest, 2022-02-28:
Thanks for your message. Can you give some more information about the timing you observe? Is there a jitter in the timestamps within the ROS-messages, or in the timestamps of their publishing? Which lidar do you use?

Please note, that the lidar and ROS system use independant clocks. Lidar timestamps are converted to system time using a software pll, which estimates the mapping between lidar timestamps and system clock from the timestamps observed whenever receiving a lidar telegram. In case of network delays, the system timestamps may vary.

rostest, 2022-07-19:
Closed due to inactivity

closed 2022-07-19 by rostest