SICKAG/sick_scan #168: MRS1000 Connection problem with ROS

omerdurmus61, 2023-03-02:
Hi, I am trying to use MRS1000 LIDAR with ROS Melodic. I am using this repository as ROS driver. I can get PointCLoud from LIDAR on SOPAS-ET software. . I connected the LIDAR to PC  via ethernet cable. When i tried to use the LIDAR on ubuntu 18.04 i set a static IP from settings->network->wired settings->IPv4->manual. After IP configuration I changed "hostname" parameter with my IP from launcher file. I got some errors when i run this command "roslaunch sick_scan sick_mrs_1xxx.launch". First of all the LIDAR stopped working. There was no sound (originating from the motor) from LIDAR. After that i got an error like below. After about a minute the LIDAR worked and i was able to get PointCLoud from topics.  The LED's on the LIDAR are always red during this procces. I think all of these and the output is not normal for the LIDAR. Maybe this error is related to PLL. How can i fix this output? 

Thanks for your held and advice.

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /sick_mrs_1xxx/cloud_topic: cloud
 * /sick_mrs_1xxx/frame_id: cloud
 * /sick_mrs_1xxx/hostname: 169.254.210.176
 * /sick_mrs_1xxx/intensity_resolution_16bit: False
 * /sick_mrs_1xxx/max_ang: 2.3998277
 * /sick_mrs_1xxx/min_ang: -2.3998277
 * /sick_mrs_1xxx/min_intensity: 0.0
 * /sick_mrs_1xxx/port: 2112
 * /sick_mrs_1xxx/scanner_type: sick_mrs_1xxx
 * /sick_mrs_1xxx/sw_pll_only_publish: True
 * /sick_mrs_1xxx/timelimit: 5

NODES
  /
    sick_mrs_1xxx (sick_scan/sick_generic_caller)

auto-starting new master
process[master]: started with pid [9138]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 75642072-b887-11ed-b7e8-1063c839b2b1
process[rosout-1]: started with pid [9156]
started core service [/rosout]
process[sick_mrs_1xxx-2]: started with pid [9159]
[ INFO] [1677712746.355909896]: sick_generic_caller V. 1.12.1
[ INFO] [1677712746.356159268]: Program arguments: /home/omer/sick_ws/devel/lib/sick_scan/sick_generic_caller
[ INFO] [1677712746.356191594]: Program arguments: __name:=sick_mrs_1xxx
[ INFO] [1677712746.356209087]: Program arguments: __log:=/home/omer/.ros/log/75642072-b887-11ed-b7e8-1063c839b2b1/sick_mrs_1xxx-2.log
[ INFO] [1677712746.402732625]: Start initialising scanner [Ip: 169.254.210.176] [Port: 2112]
[ INFO] [1677712746.503684173]: Publishing laserscan-pointcloud2 to cloud
[ INFO] [1677712746.534694183]: Parameter setting for <active_echo: 0>
[ INFO] [1677712746.738815807]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
[ERROR] [1677712752.324726453]: no answer received after 5000 ms. Maybe sopas mode is wrong.

[ INFO] [1677712752.324875251]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1677712752.325096468]: Receiving: <STX><ETX>
[ INFO] [1677712752.325142404]: SOPAS Communication -Error unexpected Sopas Answer for request <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>

[ INFO] [1677712752.325372993]: Sending  : <STX>sMN SetAccessMode 3 F4724744<ETX>
[ INFO] [1677712752.326613729]: Receiving: <STX>sAN SetAccessMode 1<ETX>
[ INFO] [1677712752.526922325]: Sending  : <STX>sWN EIHstCola 1<ETX>
[ INFO] [1677712752.528310754]: Receiving: <STX>sWA EIHstCola<ETX>
[ INFO] [1677712752.528493124]: Failed to init scanner Error Code: 1
Waiting for timeout...
If the communication mode set in the scanner memory is different from that used by the driver, the scanner's communication mode is changed.
This requires a restart of the TCP-IP connection, which can extend the start time by up to 30 seconds. There are two ways to prevent this:
1. [Recommended] Set the communication mode with the SOPAS ET software to binary and save this setting in the scanner's EEPROM.
2. Use the parameter "use_binary_protocol" to overwrite the default settings of the driver.
[ INFO] [1677712752.529825343]: Start initialising scanner [Ip: 169.254.210.176] [Port: 2112]
[ WARN] [1677712752.529945316]: Disconnecting TCP-Connection.
1677712753.5288 Info: Tcp::readInputData: Read 0 bytes - connection is lost!
sick_scan driver exiting.
[ INFO] [1677712753.611058357]: Publishing laserscan-pointcloud2 to cloud
[ INFO] [1677712753.618120464]: Parameter setting for <active_echo: 0>
[ INFO] [1677712753.818770860]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
[ INFO] [1677712753.820100513]: Receiving: <STX>sAN SetAccessMode \x01<ETX>
[ INFO] [1677712754.020570475]: Sending  : <STX><STX><STX><STX><Len=0015>sWN EIHstCola 0x01 CRC:<0x09>
[ INFO] [1677712754.021828503]: Receiving: <STX>sWA EIHstCola <ETX>
[ INFO] [1677712754.222240106]: Sending  : <STX><STX><STX><STX><Len=0015>sMN LMCstopmeas CRC:<0x10>
[ INFO] [1677712754.243096889]: Receiving: <STX>sAN LMCstopmeas \x00<ETX>
[... 110 lines cut ...]
[ INFO] [1677712911.420555193]: Waiting for scanner ready state since 20 secs
[ INFO] [1677712912.420988478]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[ INFO] [1677712912.422372416]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[ INFO] [1677712912.422512005]: Waiting for scanner ready state since 21 secs
[ INFO] [1677712913.422760345]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[ INFO] [1677712913.423978354]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[ INFO] [1677712913.424033899]: Waiting for scanner ready state since 22 secs
[ INFO] [1677712914.424394012]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[ INFO] [1677712914.425774520]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[ INFO] [1677712914.425919137]: Waiting for scanner ready state since 23 secs
[ INFO] [1677712915.426246878]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[ INFO] [1677712915.427495442]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[ INFO] [1677712915.427553169]: Waiting for scanner ready state since 24 secs
[ INFO] [1677712916.427906339]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[ INFO] [1677712916.429369782]: Receiving: <STX>sRA SCdevicestate \x01<ETX>
[ INFO] [1677712916.429579533]: Scanner ready for measurement after 25 [sec]
[ INFO] [1677712916.429896933]: Sending  : <STX><STX><STX><STX><Len=0017>sEN LMDscandata 0x01 CRC:<0x33>
[ INFO] [1677712916.431182428]: Receiving: <STX>sEA LMDscandata \x01<ETX>
[ INFO] [1677712916.449885581]: 1 / 6 Packet dropped Software PLL not yet locked.
[ INFO] [1677712916.469185072]: 2 / 6 Packet dropped Software PLL not yet locked.
[ INFO] [1677712916.489379076]: 3 / 6 Packet dropped Software PLL not yet locked.
[ INFO] [1677712916.508842005]: 4 / 6 Packet dropped Software PLL not yet locked.
[ INFO] [1677712916.528956256]: 5 / 6 Packet dropped Software PLL not yet locked.
[ INFO] [1677712916.548932293]: 6 / 6 Packet dropped Software PLL not yet locked.
[ INFO] [1677712916.548981438]: Software PLL is expected to be ready now!

BeSeeTek, 2023-03-02:
Have you configured the scanner in sopasET before? Using ASCI command mode?
It looks like that.
Does this happen all the time? 

Any ways sick_scan_xd is the recommended driver instead of sick_scan https://github.com/SICKAG/sick_scan_xd

omerdurmus61, 2023-03-02:
> Have you configured the scanner in sopasET before? Using ASCI command mode? It looks like that. Does this happen all the time?
> 
> Any ways sick_scan_xd is the recommended driver instead of sick_scan https://github.com/SICKAG/sick_scan_xd

I just configured the IP. Should i use ASCI command mode? And how can i cahange it on SOPAS software.